dusty-nv/jetson-containers #2: Run locally

g30ba1, 2020-05-04:
I've built all the containers successfully.

But I've NO idea how to run them.

Any tip?

dusty-nv, 2020-05-21:
Hi @g30ba1 , you can find the run instructions on NGC.  For example:

https://ngc.nvidia.com/catalog/containers/nvidia:l4t-ml

```bash
sudo docker run -it --rm --runtime nvidia --network host nvcr.io/nvidia/l4t-ml:r32.4.2-py3
```

uersoy, 2020-05-30:
> Hi @g30ba1 , you can find the run instructions on NGC. For example:
> 
> https://ngc.nvidia.com/catalog/containers/nvidia:l4t-ml
> 
> ```shell
> sudo docker run -it --rm --runtime nvidia --network host nvcr.io/nvidia/l4t-ml:r32.4.2-py3
> ```

Hello @dusty-nv,

I can run the container at localhost:8888 however all my saved work gets lost the next time I access the same. Seems like no checkpoint is saved when I save my jupyterlab files. Do you know why this may be happening?

Thanks!

g30ba1, 2020-05-30:
> Hi @g30ba1 , you can find the run instructions on NGC. For example:
> 
> https://ngc.nvidia.com/catalog/containers/nvidia:l4t-ml
> 
> ```shell
> sudo docker run -it --rm --runtime nvidia --network host nvcr.io/nvidia/l4t-ml:r32.4.2-py3
> ```

Hi Dusty, thank you for your time.

I found that if we re-built the containers, the command to run is:

```sudo docker run -it --rm --runtime nvidia --network host -v /home/user/Documents/Projects:/home l4t-tensorflow:r32.4.2-tf1.15-py3```

(-v flag indicates that we will be working on a local folder)

NGC containers are a nice tool to get an environment as quickly as possible.

g30ba1, 2020-05-30:
> Hello @dusty-nv,
>
> I can run the container at localhost:8888 however all my saved work gets lost the next time I access the same. Seems like no checkpoint is saved when I save my jupyterlab files. Do [...]

You must use the -v flag to WORK on a local folder while using the container, i.e.:

```sudo docker run -it --rm --runtime nvidia --network host -v /home/user/Documents/Projects:/home l4t-tensorflow:r32.4.2-tf1.15-py3```

If you are suing the container published on Nvidia´s NGC, the command is:

```sudo docker run -it --rm --runtime nvidia --network host -v /home/user/project:/location/in/container nvcr.io/nvidia/l4t-ml:r32.4.2-py3```

To use a local folder, you can use any path existent on your local machine (host), or you can create the folder while you´re INSIDE the container.

uersoy, 2020-05-31:
> You must use the -v flag to WORK on a local folder while using the container, i.e.:
>
> `sudo docker run -it --rm --runtime nvidia --network host -v /home/user/Documents/Projects:/home [...]

Thank you Jorge. I can use a local folder to save my scripts, notebooks however I am wondering how to keep the changes I make to the whole container image. Let's say I install new dependencies, libraries on top of existing default ones. Will I have to create a new container image?

uersoy, 2020-06-01:
> Thank you Jorge. I can use a local folder to save my scripts, notebooks however I am wondering how to keep the changes I make to the whole container image. Let's say I install new dependencies, [...]

I figured out. The below link explains it.
https://docs.nvidia.com/deeplearning/frameworks/user-guide/index.html

g30ba1, 2020-06-01:
Nice find! I'm already reading it.

Regarding your question, yes, you have to re-built an image, to add libraries, dependencies or any other requirements for your projects.

uersoy, 2020-06-01:
> Nice find! I'm already reading it.
> 
> Regarding your question, yes, you have to re-built an image, to add libraries, dependencies or any other requirements for your projects.

I also read the whole thing to learn but I found the answer to my question in Section 10.1.4.

closed 2025-02-21 by johnnynunez