dusty-nv/jetson-containers #4: Cameras not listed in /dev/video*

astekardis, 2020-05-13:
Hey @dusty-nv,

Thanks for putting this together! I have a Jetson Nano running L4T version 32.3.1, so, for testing purposes, I pulled the L4T base image called `nvcr.io/nvidia/l4t-base:r32.3.1`. 

I start the image with the  command `sudo docker run -it --rm --net=host --runtime nvidia  -e DISPLAY=$DISPLAY -v /tmp/.X11-unix/:/tmp/.X11-unix nvcr.io/nvidia/l4t-base:r32.3.1`. 

I have one USB 2.0 camera connected to the device, and when I run the command `ls /dev/video*` in the image, the output is `ls: cannot access '/dev/video*': No such file or directory`. However, if I run this command on the host OS, the output is as follows: `/dev/video0  /dev/video1`

Why is the image not able to see the camera at this location? I assume that there may be a package missing from the base image that needs to be installed. Let me know if this isn't the right place for such a question and I'll move it to the appropriate location.

dusty-nv, 2020-05-13:
Hi @astekardis, try using `--device /dev/video0` and  `--device /dev/video1` in your `docker run` command.

astekardis, 2020-05-13:
Thanks, that worked.

closed 2020-05-13 by astekardis