dusty-nv/jetson-containers #5: OpenCV and nvidia-l4t-base

astekardis, 2020-05-21:
Hey @dusty-nv, I'm new to using docker and I'm trying to figure out the best way to include OpenCV in my docker image that I am building on top of `nvidia-l4t-base`. In order to reduce the overall size of the image, I think the best way to do this is to use the opencv installation that is installed when the Jetson is flashed. I think this is the same general approach that you take to keep `nvidia-l4t-base` relatively small - you use the host's CUDA installation rather than copying it into the docker image. 

Is it possible to do the same with the host's OpenCV installation? If so, could you point me to any relevant docs that could show me how to do this?

dusty-nv, 2020-05-21:
Hi @astekardis , here are the docs for the mount files:  https://github.com/NVIDIA/libnvidia-container/blob/jetson/design/mount_plugins.md

You can find them under `/etc/nvidia-container-runtime/host-files-for-container.d`.  But yes, it may just be easier to install opencv into your container with apt.

astekardis, 2020-05-21:
Okay, thanks for that info. That seems like a pretty involved process. Once I create the appropriate csv, will it automatically be used within the docker image, or is there a specific deployment process that I have to follow?

Yeah, apt installing is certainly the easier way to do this, but features would be quite different, right? I believe that when you apt install opencv, you get version 3.2.0, whereas opencv 4.1.1 is what is installed by default on Jetson devices.

dusty-nv, 2020-05-21:
> Once I create the appropriate csv, will it automatically be used within the docker image, or is there a specific deployment process that I have to follow?

I think you need to reboot or restart your docker service after modifying the CSV's, but yes after that it should show up in your containers (as long as they were started with `--runtime nvidia`)

> Yeah, apt installing is certainly the easier way to do this, but features would be quite different, right? I believe that when you apt install opencv, you get version 3.2.0, whereas opencv 4.1.1 is what is installed by default on Jetson devices.

I haven't tried this, but you could in theory copy the `/etc/apt/sources.list.d/nvidia-l4t-apt-source.list/` file from your Jetson device into your container, and then you would be able to pull the NVIDIA JetPack packages from apt.

astekardis, 2020-05-21:
Hmm I'm trying this out and am having some issues. I feel like I'm probably missing something basic here, but I created `/etc/apt/sources.list.d/nvidia-l4t-apt-source.list` in my container and populated it with the info from my device. I ran `sudo apt update` and got the following errors:

```
W: GPG error: https://repo.download.nvidia.com/jetson/common r32 InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 0D296FFB880FB004
E: The repository 'https://repo.download.nvidia.com/jetson/common r32 InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: GPG error: https://repo.download.nvidia.com/jetson/t210 r32 InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 0D296FFB880FB004
E: The repository 'https://repo.download.nvidia.com/jetson/t210 r32 InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Trying to get around this, I ran `sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 0D296FFB880FB004`, which yielded the following output:
```
Executing: /tmp/apt-key-gpghome.YWecuNEBSM/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 0D296FFB880FB004
gpg: keyserver receive failed: No data
```
Did this happen because I'm using the wrong keyserver?

rdejana, 2020-05-22:
I copied the gpg key from my host (tx2).

astekardis, 2020-05-22:
@rdejana You copied it into a docker container? Can you provide more details on what you did?

rdejana, 2020-05-22:
Yea, did this.  Created a working directory for building my container and cd'ed into.  Then
```
  cp /etc/apt/sources.list.d/nvidia-l4t-apt-source.list .
  cp /etc/apt/trusted.gpg.d/jetson-ota-public.asc .
```
Then my dockerfile looks like
```
FROM  nvcr.io/nvidia/l4t-base:r32.4.2
ENV DEBIAN_FRONTEND=noninteractive
# needed for accessing jetpack.  This is for 4.4
COPY  nvidia-l4t-apt-source.list /etc/apt/sources.list.d/nvidia-l4t-apt-source.list
COPY  jetson-ota-public.asc /etc/apt/trusted.gpg.d/jetson-ota-public.asc
RUN apt-get update

RUN apt-get update && apt-get install -y libopencv-python && apt-get install -y --no-install-recommends \
          python3-pip \
          python3-dev \
          build-essential \
          zlib1g-dev \
          zip \
          libjpeg8-dev && rm -rf /var/lib/apt/lists/*

RUN pip3 install setuptools Cython wheel
RUN pip3 install numpy --verbose

```

astekardis, 2020-05-22:
> # needed for accessing jetpack. This is for 4.4
Does this mean that you can only copy the source list and public key for Jetpack 4.4 devices? Is it possible to do something similar for devices running older jetpack versions?

rdejana, 2020-05-22:
I haven't tried anything older, sorry.  It should work, just don't have anything to try it out with.

astekardis, 2020-05-22:
Okay, thanks for the info so far. Just out of curiosity, what version of opencv is installed in your docker image?

drakorg, 2020-06-30:
> Okay, thanks for the info so far. Just out of curiosity, what version of opencv is installed in your docker image?

@astekardis Hi, I'm currently compiling opencv inside the container as I write this, but I would very much like to try using the host's opencv libraries in the same fashion as other dependencies are handled by nvidia, i.e. by supplying a custom opencv.csv file to the host-files-for-container.d directory. Were you able to tailor such file? It seems that most of opencv files are under /usr/lib/aarch64-linux-gnu/ but how to tell if I'm not missing anything. I've been doing nothing but guessing so far. Educated guessing at best.

Thank you,
Best regards.
Eduardo

andreasberg85, 2020-08-13:
Hi,

My host didn't have the file `/etc/apt/trusted.gpg.d/jetson-ota-public.asc`.

But I added this line to my `Dockerfile`:
`RUN apt-key adv --fetch-key https://repo.download.nvidia.com/jetson/jetson-ota-public.asc`

Seemed to work! Host is Xavier, running JP4.3. I'm now able to run a  Python script with `import cv2`. Dockerfile builds from this image: `stereolabs/zed:3.2-py-devel-jetson-jp4.3`

Thanks guys! BR Andreas

mdegans, 2020-11-24:
> Hi,
> 
> My host didn't have the file `/etc/apt/trusted.gpg.d/jetson-ota-public.asc`.
> 
> But I added this line to my `Dockerfile`:
> `RUN apt-key adv --fetch-key https://repo.download.nvidia.com/jetson/jetson-ota-public.asc`
> 
> Seemed to work! Host is Xavier, running JP4.3. I'm now able to run a Python script with `import cv2`. Dockerfile builds from this image: `stereolabs/zed:3.2-py-devel-jetson-jp4.3`.
> 
> Otherwise, I basically follow the Dockerfile of @rdejana
> 
> Thanks guys! BR Andreas

Thanks a lot for that, @andreasberg85 

Before, I was downloading a whole bsp tarball just to get at that file (it's not there anymore). The url to the key is exactly what I needed.

umairjavaid, 2021-09-25:
any easy solutions for this?

Tibalt, 2022-03-18:
an example of opencv.csv

```
nvidia@miivii-tegra:~$ sudo vi /etc/nvidia-container-runtime/host-files-for-container.d/opencv.csv
lib, /usr/lib/aarch64-linux-gnu/libopencv_imgcodecs.so
lib, /usr/lib/aarch64-linux-gnu/libopencv_imgcodecs.so.4.1
lib, /usr/lib/aarch64-linux-gnu/libopencv_imgcodecs.so.4.1.1
lib, /usr/lib/aarch64-linux-gnu/libopencv_core.so
lib, /usr/lib/aarch64-linux-gnu/libopencv_core.so.4.1
lib, /usr/lib/aarch64-linux-gnu/libopencv_core.so.4.1.1
lib, /usr/lib/aarch64-linux-gnu/libopencv_imgproc.so
lib, /usr/lib/aarch64-linux-gnu/libopencv_imgproc.so.4.1
lib, /usr/lib/aarch64-linux-gnu/libopencv_imgproc.so.4.1.1
```

Then OpenCV libraries could be mapped into docker and be used without problem. I think  it is easier than installing it in an image.

closed 2025-02-21 by johnnynunez