dusty-nv/jetson-containers #8: can't run latest release

makslevental, 2020-05-28:
```
max@jetson-tx2-0:~$ sudo docker run -it --rm --runtime nvidia --network host nvcr.io/nvidia/l4t-pytorch:r32.4.2-pth1.5-py3
docker: Error response from daemon: OCI runtime create failed: container_linux.go:349: starting container process caused "process_linux.go:449: container init caused \"process_linux.go:432: running prestart hook 0 caused \\\"error running hook: exit status 1, stdout: , stderr: exec command: [/usr/bin/nvidia-container-cli --load-kmods configure --ldconfig=@/sbin/ldconfig.real --device=all --compute --compat32 --graphics --utility --video --display --pid=15566 /var/lib/docker/overlay2/d7c28eeff71d3aebfac3cce5ded7bd61a894f1f2d4922bf1966d993f2da86fc0/merged]\\\\nnvidia-container-cli: mount error: file creation failed: /var/lib/docker/overlay2/d7c28eeff71d3aebfac3cce5ded7bd61a894f1f2d4922bf1966d993f2da86fc0/merged/usr/lib/libvisionworks.so: file exists\\\\n\\\"\"": unknown.
```

dusty-nv, 2020-05-28:
Are you on JetPack 4.4 Developer Preview?

makslevental, 2020-05-28:
@dusty-nv sorry it turns out that in fact i am not

```
max@jetson-tx2-0:~$ sudo apt-cache show nvidia-jetpack
Package: nvidia-jetpack
Version: 4.3-b134
Architecture: arm64
Maintainer: NVIDIA Corporation
Installed-Size: 194
Depends: nvidia-container-csv-cuda (= 10.0.326-1), libopencv-python (= 4.1.1-2-gd5a58aa75), libvisionworks-sfm-dev (= 0.90.4), libvisionworks-dev (= 1.6.0.500n), libvisionworks-samples (= 1.6.0.500n), libnvparsers6 (= 6.0.1-1+cuda10.0), libnvinfer-plugin6 (= 6.0.1-1+cuda10.0), libcudnn7-doc (= 7.6.3.28-1+cuda10.0), libcudnn7-dev (= 7.6.3.28-1+cuda10.0), libnvinfer-samples (= 6.0.1-1+cuda10.0), libnvinfer-bin (= 6.0.1-1+cuda10.0), nvidia-container-csv-cudnn (= 7.6.3.28-1+cuda10.0), libvisionworks-tracking-dev (= 0.88.2), vpi-samples (= 0.1.0), tensorrt (= 6.0.1.10-1+cuda10.0), libopencv (= 4.1.1-2-gd5a58aa75), libnvinfer-doc (= 6.0.1-1+cuda10.0), libnvparsers-dev (= 6.0.1-1+cuda10.0), libcudnn7 (= 7.6.3.28-1+cuda10.0), libnvidia-container0 (= 0.9.0~beta.1), cuda-toolkit-10-0 (= 10.0.326-1), nvidia-container-csv-visionworks (= 1.6.0.500n), graphsurgeon-tf (= 6.0.1-1+cuda10.0), libopencv-samples (= 4.1.1-2-gd5a58aa75), python-libnvinfer-dev (= 6.0.1-1+cuda10.0), libnvinfer-plugin-dev (= 6.0.1-1+cuda10.0), libvisionworks (= 1.6.0.500n), nvidia-container-toolkit (= 1.0.1-1), libnvinfer-dev (= 6.0.1-1+cuda10.0), nvidia-l4t-jetson-multimedia-api (= 32.3.1-20191209230245), libopencv-dev (= 4.1.1-2-gd5a58aa75), vpi-dev (= 0.1.0), vpi (= 0.1.0), python3-libnvinfer (= 6.0.1-1+cuda10.0), python3-libnvinfer-dev (= 6.0.1-1+cuda10.0), opencv-licenses (= 4.1.1-2-gd5a58aa75), nvidia-container-csv-tensorrt (= 6.0.1.10-1+cuda10.0), libnvinfer6 (= 6.0.1-1+cuda10.0), libnvonnxparsers-dev (= 6.0.1-1+cuda10.0), libnvonnxparsers6 (= 6.0.1-1+cuda10.0), uff-converter-tf (= 6.0.1-1+cuda10.0), nvidia-docker2 (= 2.2.0-1), libvisionworks-sfm (= 0.90.4), libnvidia-container-tools (= 0.9.0~beta.1), nvidia-container-runtime (= 3.1.0-1), python-libnvinfer (= 6.0.1-1+cuda10.0), libvisionworks-tracking (= 0.88.2)
Homepage: http://developer.nvidia.com/jetson
Priority: standard
Section: metapackages
Filename: pool/main/n/nvidia-jetpack/nvidia-jetpack_4.3-b134_arm64.deb
Size: 29746
SHA256: 958c32d872e11fef50737f6f38f93dd2c737f5d68886cb964d3da63151405337
SHA1: 32aa5a29be86550fbe9f4c249f2640bdfce596cd
MD5sum: 2b2590855299af3030509e55d9135c2d
Description: NVIDIA Jetpack Meta Package
Description-md5: ad1462289bdbc54909ae109d1d32c0a8
```
let me see if upgrading solves

closed 2020-05-28 by makslevental