dusty-nv/jetson-containers #10: build pytorch c++ api from this image?

MoonBlvd, 2020-05-31:
Hi, I would like to build pytorch c++ api based upon your pytorch 1.5 image. However it seems that pytorch in the image is built from wheel file so that I cannot build c++ api from source. Moreover, it seems that this image doesn't have cudnn and its header files. Could you please give me some suggestions about how to build pytorch c++ api in your image? Thank you!

dusty-nv, 2020-06-01:
> Moreover, it seems that this image doesn't have cudnn and its header files.

Hi @MoonBlvd, the CUDA/cuDNN/ect header files and libraries are automatically mapped from your host Jetson device by the nvidia docker runtime.   You should set the `"default-runtime": "nvidia"` as shown here:  https://github.com/dusty-nv/jetson-containers#docker-default-runtime

> However it seems that pytorch in the image is built from wheel file so that I cannot build c++ api from source.

It was built with libtorch and you should find the pytorch C/C++ headers under it's python install dir.  But if you need to build PyTorch from source, see here:  https://forums.developer.nvidia.com/t/pytorch-for-jetson-nano-version-1-5-0-now-available/

MoonBlvd, 2020-06-01:
Hi @dusty-nv Thank you so much for your response! I have added `"default-runtime": "nvidia"` to the daemon.json as you suggested and I used `--runtime nvidia` when running `sudo docker run`. However, I still cannot find those cuDNN headerfiles or lib files (`.so` files I guess?) inside the container. 

FYI, I have JetPack 4.4 on my TX2 and I have all cuDNN header files locate in `/usr/include`. It seems that these files are links, e.g. `lrwxrwxrwx 1 root root 26 May 17 01:24 /usr/include/cudnn.h -> /etc/alternatives/libcudnn`.  I can also find library files like `libcudnn.so` in `/usr/lib/aarch64-linux-gnu`

MoonBlvd, 2020-06-01:
This is a follow up of my questions in #3

dusty-nv, 2020-06-01:
> However, I still cannot find those cuDNN headerfiles or lib files (`.so` files I guess?) inside the container.

Sorry about that, I am checking with our Docker team about why that is happening.

As a temporary workaround, you could manually copy the files you need into your docker workspace, and then copy these files into your image in your dockerfile.

MoonBlvd, 2020-06-03:
@dusty-nv Thank you for your response! Please let me know if you got any solutions!

dusty-nv, 2020-06-04:
Following up on this - run the following:

```bash
$ sudo apt-get install dos2unix
$ sudo dos2unix  /etc/nvidia-container-runtime/host-files-for-container.d/cudnn.csv 
```
Then reboot or restart your docker service, and you should then see the cudnn libraries/headers in the containers.

MoonBlvd, 2020-06-04:
@dusty-nv  Thank you! Your solution works for me! So it seems that the container runtime was not able to map these host files because this csv files was not converted to unix readable format, it that correct? 

Can I ask another question about building pytorch and torchvision in directly on the device? I followed your dockerfile for pytorch to install depencencies and pytorch-1.5.0. Everything went well until I started to build torchvision-v0.4.0.  I got the following errors:

```
In file included from /home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/models.h:6:0,
                 from /home/brianyao_tx2/torchvision/test/test_models.cpp:5:
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/googlenet.h:13:14: error: ‘BatchNorm’ in namespace ‘torch::nn’ does not name a type
   torch::nn::BatchNorm bn{nullptr};
              ^~~~~~~~~
In file included from /home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/models.h:7:0,
                 from /home/brianyao_tx2/torchvision/test/test_models.cpp:5:
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/inception.h:12:14: error: ‘BatchNorm’ in namespace ‘torch::nn’ does not name a type
   torch::nn::BatchNorm bn{nullptr};
              ^~~~~~~~~
In file included from /home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/models.h:10:0,
                 from /home/brianyao_tx2/torchvision/test/test_models.cpp:5:
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:31:14: error: ‘BatchNorm’ in namespace ‘torch::nn’ does not name a type
   torch::nn::BatchNorm bn1{nullptr}, bn2{nullptr};
              ^~~~~~~~~
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:31:36: error: expected unqualified-id before ‘,’ token
   torch::nn::BatchNorm bn1{nullptr}, bn2{nullptr};
                                    ^
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:54:14: error: ‘BatchNorm’ in namespace ‘torch::nn’ does not name a type
   torch::nn::BatchNorm bn1{nullptr}, bn2{nullptr}, bn3{nullptr};
              ^~~~~~~~~
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:54:36: error: expected unqualified-id before ‘,’ token
   torch::nn::BatchNorm bn1{nullptr}, bn2{nullptr}, bn3{nullptr};
                                    ^
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:54:50: error: expected unqualified-id before ‘,’ token
   torch::nn::BatchNorm bn1{nullptr}, bn2{nullptr}, bn3{nullptr};
                                                  ^
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:74:14: error: ‘BatchNorm’ in namespace ‘torch::nn’ does not name a type
   torch::nn::BatchNorm bn1;
              ^~~~~~~~~
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h: In member function ‘torch::nn::Sequential vision::models::ResNetImpl<Block>::_make_layer(int64_t, int64_t, int64_t)’:
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:102:20: error: ‘BatchNorm’ is not a member of ‘torch::nn’
         torch::nn::BatchNorm(planes * Block::expansion));
                    ^~~~~~~~~
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:102:20: note: suggested alternative: ‘BatchNorm3d’
         torch::nn::BatchNorm(planes * Block::expansion));
                    ^~~~~~~~~
                    BatchNorm3d
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h: In constructor ‘vision::models::ResNetImpl<Block>::ResNetImpl(const std::vector<int, std::allocator<int> >&, int64_t, bool, int64_t, int64_t)’:
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:127:69: error: ‘struct torch::nn::ConvOptions<2>’ has no member named ‘with_bias’
       conv1(torch::nn::Conv2dOptions(3, 64, 7).stride(2).padding(3).with_bias(
                                                                     ^~~~~~~~~
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:129:7: error: class ‘vision::models::ResNetImpl<Block>’ does not have any field named ‘bn1’
       bn1(64),
       ^~~
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:136:26: error: ‘bn1’ was not declared in this scope
   register_module("bn1", bn1);
                          ^~~
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:149:28: error: ‘torch::nn::init::FanMode’ has not been declared
           torch::nn::init::FanMode::FanOut,
                            ^~~~~~~
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:150:28: error: ‘torch::nn::init::Nonlinearity’ has not been declared
           torch::nn::init::Nonlinearity::ReLU);
                            ^~~~~~~~~~~~
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:151:47: error: ‘BatchNormImpl’ in namespace ‘torch::nn’ does not name a type
     else if (auto M = dynamic_cast<torch::nn::BatchNormImpl*>(module.get())) {
                                               ^~~~~~~~~~~~~
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:151:60: error: expected ‘>’ before ‘*’ token
     else if (auto M = dynamic_cast<torch::nn::BatchNormImpl*>(module.get())) {
                                                            ^
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:151:60: error: expected ‘(’ before ‘*’ token
[... cut ...]
         torch::nn::init::constant_(M->bn2->weight, 0);
                                       ^~~
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h: In member function ‘at::Tensor vision::models::ResNetImpl<Block>::forward(at::Tensor)’:
/home/brianyao_tx2/torchvision/test/../torchvision/csrc/models/resnet.h:173:7: error: ‘bn1’ was not declared in this scope
   x = bn1->forward(x).relu_();
       ^~~
error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
```
Do you know what might be the problem? Thank you!

MoonBlvd, 2020-06-04:
Well the `torchvision` problem was solved. 
Pytorch 1.5.0 is not compatible with torchvision v0.4.0. In stead, I have to install v0.6.0. Would be great if you could add this to the comments in `Dockerfile.pytorch`, just in case other people might want to follow your dockerfile but with a different version of pytorch.

closed 2025-02-21 by johnnynunez