dusty-nv/jetson-containers #15: I can't create a new file, the file system is reported as read-only

land007, 2020-08-18:
Step 16/19 : RUN cd /usr/local/cuda/targets/aarch64-linux/lib && mkdir 123
 ---> Running in d0f4ef0fa4c4
mkdir: cannot create directory '123': Read-only file system

dusty-nv, 2020-08-18:
`/usr/local/cuda` is mounted from the host device in read-only mode, so you can't write to it from within the containers.

For more info, please see here:   https://github.com/NVIDIA/nvidia-docker/wiki/NVIDIA-Container-Runtime-on-Jetson#mount-plugins

land007, 2020-08-18:
Currently I have the following version environment
 -Libraries:
    * CUDA: 10.2.89
    * cuDNN: 8.0.0.145
    * TensorRT: 7.1.0.16
I hope to make a docker image of the following environment
  -Libraries:
    * CUDA: 10.0
    * cuDNN: 7.5
    * TensorRT: 5.1.6
If docker depends on the environment of the host, is it impossible to complete such a mirror

dusty-nv, 2020-08-18:
It is currently not supported to use a different version of CUDA/ect than what comes with JetPack-L4T.

However you could try making your own base image and install the CUDA packages inside your container.  What you should do is move/rename `/etc/nvidia-container-runtime/host-files-for-container.d/cuda.csv` on your host device first (and `cudnn.csv`, `tensorrt.csv`), so those CUDA files don't get mounted into your container.

land007, 2020-08-18:
Thanks for your reply, it does not seem easy. The production of nvcr.io/nvidia/l4t-base does not seem to be in this project. I am not sure if the host cuda driver works for the low version cuda calls in Docker.

land007, 2020-08-26:
I finally understand what you mean. After repeated attempts, I achieved the downgrade of CUDA.[link](https://github.com/tensorflow/tfjs/issues/2872#issuecomment-680728565)

closed 2025-02-21 by johnnynunez