dusty-nv/jetson-containers #22: How to run ROS2 foxy container and source ros

kylelscott, 2020-10-14:
Hello dusty_nv. Thank you for creating these containers and making this tutorial. I used it to run ros2 FOXY, and I was able to download and test the container and it passed the test. However, I am new to docker and do not know how to run the container so I can source Foxy and run ROS2 commands. Also, do I have to worry about my workspace getting overriden/not saved when I stop the container, or if I run the container will I be able to work in my ws and when I stop the container the ws will still be saved?

dusty-nv, 2020-10-14:
> However, I am new to docker and do not know how to run the container so I can source Foxy and run ROS2 commands.

Hi @kylescott, you can run the foxy container like so:

``` bash
$ sudo docker run -runtime nvidia -it --rm --network host ros:foxy-ros-base-l4t-r32.4.3
```

The ROS2 environment should be automatically sourced when the container is started, so you shouldn't have to worry about that.

> Also, do I have to worry about my workspace getting overriden/not saved when I stop the container, or if I run the container will I be able to work in my ws and when I stop the container the ws will still be saved?

If you stop the container, then any changes that you made inside the container will be lost.   However if you run `sudo docker commit` from your host device before the container is stopped, then you can save any changes to a new container.  For more info, see here:

https://docs.docker.com/engine/reference/commandline/commit/

kylelscott, 2020-10-14:
Hello @dusty-nv. II'm getting this error for a -r flag, shoudl the command say "-runtime" or "runtime"?

`sudo docker run -runtime nvidia -it --rm --network host ros:foxy-ros-base-l4t-r32.4.3
unknown shorthand flag: 'r' in -runtime`

dusty-nv, 2020-10-14:
> II'm getting this error for a -r flag, shoudl the command say "-runtime" or "runtime"?

Sorry, yes it should be `--runtime`, like below:

``` bash
$ sudo docker run --runtime nvidia -it --rm --network host ros:foxy-ros-base-l4t-r32.4.3
```

kylelscott, 2020-10-14:
Awesome, this worked. Thank you

closed 2025-02-21 by johnnynunez