dusty-nv/jetson-containers #23: System reboot when colcon build

stanislavkuskov, 2020-10-19:
Hi! my problem is Jetson will reboot when colcon build ros foxy (desktop).
Maybe you know why?

(trying in power mode 0, 3, and 7)

dusty-nv, 2020-10-19:
Hmm, which Jetson are you using?   Have you seen it reboot like that before?   Does it happen in the minimum power mode?

is it running out of memory?   You might want to try mounting swap if it is.

stanislavkuskov, 2020-10-21:
I'm use AXG Xavier dev kit. I not see why it reboot because installing is slow and i usually don't follow. I now what it happens in the end, when rviz modules installing. I add rviz into --skip-keys and its help, but I want to understand the problem. 

I have not tried in minimal power mode. Swap is 15 gb

stanislavkuskov, 2021-08-09:
This problem was inside os with Bluetooth power management driver. I solwe it by turning off this driver

closed 2021-08-09 by stanislavkuskov