dusty-nv/jetson-containers #25: ./scripts/docker_build_ml.sh all  fails with “unauthorized: authentication required"

jvschw, 2020-10-26:
Hi Dusty,
I am trying to build the docker  container on my nano (Jetpack 4.4.1), but the process halts with the message "unauthorized: authentication required"

I cloned the repository:
git clone https://github.com/dusty-nv/jetson-containers.git

cd ed into the repository:
cd jetson-containers/

./scripts/docker_build_ml.sh all  

yields the following output:
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.4.4
l4t-base image:  nvcr.io/nvidian/nvidia-l4t-base:r32.4.4
building PyTorch torch-1.6.0-cp36-cp36m-linux_aarch64.whl, torchvision v0.7.0 (pillow), torchaudio v0.6.0
Building l4t-pytorch:r32.4.4-pth1.6-py3 container...
[sudo] password for XXXX: 
Sending build context to Docker daemon  183.3kB
Step 1/22 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.4.3
Step 2/22 : FROM ${BASE_IMAGE}
unauthorized: authentication required

Can you help me out of the ditch? I would eventually like to alter the Dockerfile in order to create my own version, but I already get stuck here. 

Best,
Jens

dusty-nv, 2020-10-26:
Hi @jvschw , sorry about that - can you try updating your jetson-containers repo?

I just fixed this is commit [`215b3c`](https://github.com/dusty-nv/jetson-containers/commit/215b3c08f20b6c69027af2d426fee079f8f6cd36)

jvschw, 2020-10-27:
Hi Dustin,
awesome, thanks for fixing this so quickly. 

My attempt to add scikit-image (by adding the line 
RUN pip3 install scikit-image --verbose
at line 93 of Dockerfile.ml) unfortunately failed, producing the error output below. I am aware that this does not pertain to the original issue, which I think is resolved, but maybe you can nevertheless help out.

Thanks again,
Jens

error message below:

 Caching due to etag
  Running setup.py (path:/tmp/pip-build-p5nhoan0/scikit-image/setup.py) egg_info for package scikit-image
    Running command python setup.py egg_info
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-p5nhoan0/scikit-image/setup.py", line 30, in <module>
        LONG_DESCRIPTION = f.read()
      File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
        return codecs.ascii_decode(input, self.errors)[0]
    UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 4029: ordinal not in range(128)
Cleaning up...
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-p5nhoan0/scikit-image/
Exception information:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/lib/python3/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 634, in _prepare_file
    abstract_dist.prep_for_dist()
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 129, in prep_for_dist
    self.req_to_install.run_egg_info()
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 439, in run_egg_info
    command_desc='python setup.py egg_info')
  File "/usr/lib/python3/dist-packages/pip/utils/__init__.py", line 725, in call_subprocess
    % (command_desc, proc.returncode, cwd))
pip.exceptions.InstallationError: Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-p5nhoan0/scikit-image/
The command '/bin/sh -c pip3 install scikit-image --verbose' returned a non-zero code: 1

closed 2025-02-21 by johnnynunez