dusty-nv/jetson-containers #26: Request: Provide TF 2.3 in ml container / Provide Jupyter in TF / Torch containers

dspyrhsu, 2020-11-02:
Hi, thanks for the excellent work here, but might I suggest you provide the ml-container with the more current tensorflow 2.3 (possibly as an option) rather that the (outdated) tensorflow 1.15?

Furthermore, I believe it would be fantastic if you could provide the additional ml packages, especially Jupyter, also in the tensorflow respectively pytorch pacakges.

Best regards,

Reinhold

dusty-nv, 2020-11-02:
Hi @dspyrhsu , thanks for the feedback!   We will have to look at the number of users on TF 1.x vs TF 2.x and determine when to switch over the l4t-ml container.

Regarding l4t-pytorch and l4t-tensorflow, we do try to keep those containers smaller (whereas there is not as big an emphasis on that in the l4t-ml container).   

All of the dockerfiles and build scripts for these containers is public though, so feel free to modify them and build your own containers:   https://github.com/dusty-nv/jetson-containers

znmeb, 2020-11-02:
I have a fork of this repo that builds two ML images, one tagged with each version of TensorFlow. It seems like you could push both to NGC and let people download either or both. There's a distressing amount of software out there that only works with 1.15, so it needs to be "always" available.

dspyrhsu, 2020-11-02:
Hi, thanks for getting back to me. I actually just built a version of l4t-ml with tf2.3 instead of tf1.15 by modifying the last line of the build script for all. In order for this to work with sklearn, I have to pass an environment variable when running the container (see here for the arising issue and solution: https://github.com/keras-team/keras-tuner/issues/317). Otherwise, I was able to do a transfer learning with densenet 121 on the Nano (with a very small data set, but still ...) - so, I am pretty impressed and thanks again.

Also, I will definitely have a look at your fork, @znmeb and I also thought about the miniforge approach. However, as of now, conda still delivers tf2.2, which is a complete disaster. So I will try that only after they ship tf2.3 or higher.

znmeb, 2020-11-02:
@dspyrhsu My conda-forge scripts are installing the NVIDIA TensorFlow and PyTorch, not the conda-forge ones. I need to do some testing to verify that they actually execute, but they do install with `pip`. Current script is here

https://github.com/znmeb/edgyR/blob/add-miniforge-arrow/docker-build-context/ml/image-installs/0miniforge

znmeb, 2020-11-05:
OK ... my fork is all tested and the images are on Docker Hub. The fork has moved: it's now at

https://github.com/edgyR/jetson-containers.

If @dusty-nv wants a pull request I'd be happy to send it along. The resulting images are pushed to 

https://hub.docker.com/u/edgyr

That organization also has my images with R, OpenCPU and RStudio Server, but I'm still cleaning up the documentation for them.

japrogramer, 2020-11-05:
I am using a jetson nano with the `nvcr.io/nvidia/l4t-tensorflow   r32.4.3-tf2.2-py3` image
because it is the only image I found with tensorflow 2 at

 https://ngc.nvidia.com/catalog/containers?orderBy=scoreDESC&pageNumber=0&query=jetson&quickFilter=containers&filters= 

but unfortunately it does not have jupyterlab So I would also like to have jupyterlab with the tensorflow Dockerfile

znmeb, 2020-11-05:
@japrogramer Try `docker pull edgyr/l4t-ml:r32.4.4-tf2.3-py3` and if you find bugs open an issue at https://github.com/edgyR/jetson-containers/issues/new

japrogramer, 2020-11-07:
@znmeb I noticed that the only change that will conflict is the change to where the built imaged is pushed.
Any chance we can get a pull request?

having tf2 in the ml images would make that set of images consistent with how the tensorflow images are tagged  at https://ngc.nvidia.com/catalog/containers/nvidia:l4t-tensorflow

znmeb, 2020-11-08:
the push change is a separate branch - the main branch has NVIDIA's push script

closed 2025-02-21 by johnnynunez