dusty-nv/jetson-containers #27: Issue when building pytorch from dockerfile

adamwhats97, 2020-11-03:
Hi there, I keep on having an issue when I try to build the pytorch container - specifically it fails when it tries to clone in torchvision and gives me the following error:

Step 15/23 : RUN git clone -b ${TORCHVISION_VERSION} https://github.com/pytorch/vision torchvision &&     cd torchvision &&     python3 setup.py install &&     cd ../ &&     rm -rf torchvision &&     pip3 install "${PILLOW_VERSION}"
 ---> Running in b6f708c93740
Cloning into 'torchvision'...
Note: checking out '78ed10cc51067f1a6bac9352831ef37a3f842784'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Traceback (most recent call last):
  File "setup.py", line 13, in <module>
    import torch
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 188, in <module>
    _load_global_deps()
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 141, in _load_global_deps
    ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libcurand.so.10: cannot open shared object file: No such file or directory
The command '/bin/sh -c git clone -b ${TORCHVISION_VERSION} https://github.com/pytorch/vision torchvision &&     cd torchvision &&     python3 setup.py install &&     cd ../ &&     rm -rf torchvision &&     pip3 install "${PILLOW_VERSION}"' returned a non-zero code: 1

I've tried remiving the -b argument but that also doesn't seem to help. I'm very new to docker so there's a chance i'm missing something obvious, but I'd really appreciate any suggestions

adamwhats97, 2020-11-03:
My bad, I hadn't set the default runtime to nvidia

closed 2020-11-03 by adamwhats97

maghauke, 2021-04-29:
I have the same error, but I have created the daemon.json in  /etc/docker/, and copied over the suggested code. I am quite new to docker, so probably a newbie mistake... Any suggestions on how to do this correctly?

dusty-nv, 2021-04-29:
> I have the same error, but I have created the daemon.json in /etc/docker/, and copied over the suggested code. I am quite new to docker, so probably a newbie mistake... Any suggestions on how to do this correctly?

Hi @maghauke, if you run `sudo docker info`, do you see `Default Runtime: nvidia` printed out?

Also, what version of JetPack-L4T are you running?   You can check with `cat /etc/nv_tegra_release`

maghauke, 2021-05-06:
Thank you for your response, under Default Runtime, nvidia is printed out.

I have version 5.1

dusty-nv, 2021-05-06:
Do you see CUDA toolkit installed on your device under `/usr/local/cuda`?

If you `ls /usr/local/cuda/lib64`, do you see `libcurand.so.10` there?