dusty-nv/jetson-containers #29: Ros foxy with ubuntu focal (20.04)

olmerg, 2020-11-06:
There is some provision to support ubuntu 20.04 with ros foxy?

dusty-nv, 2020-11-06:
JetPack is on Ubuntu 18.04, so ROS Foxy is built from source in the container so it runs on 18.04.

Myzhar, 2021-01-09:
@dusty-nv is there a date for Ubuntu 20.04 on Jetson?
Ubuntu 18.04 is a big limitation now...

dusty-nv, 2021-01-09:
Not currently, but you can build these for 18.04, even outside of the containers by following the same procedure in the dockerfiles.

mutsuyuki, 2021-04-28:
@dusty-nv 

Thank you for your great Dockerfile.
I succeeded to build foxy on Jetpack 4.4 and succeeded to run simple pub sub program.
But I try to install ros-foxy-turtlesim , I got error as below.
`
apt update
apt install ros-foxy-turtlesim
E: Unable to locate package ros-foxy-turtlesim
`
It's seem to occur at installing any ros-foxy-xxxxx.
I think the reason for this is that the apt repository is set to bionic (18.04).
I change `bionic` to `focal` in `/etc/apt/sources.list.d/ros2-latest.list` and retry to install, but I got error related to dependencies.

Is there any workaround this issue?

dusty-nv, 2021-04-28:
You need to build these packages from source as opposed to trying to install them from apt, because they aren't built for 18.04.  The container builds ROS2 Foxy core/base from source.

mutsuyuki, 2021-05-10:
@dusty-nv 

Thank you for your answer.
I understand the reason for the error.
It's hard for me to build a any package for foxy, so I'll use eloquent until the 20.04 based jetpack is released.

closed 2025-02-21 by johnnynunez