dusty-nv/jetson-containers #31: Issue when building pytorch image

aldrichg9, 2020-11-16:
When attempting to build a pytorch image I run: 
`./scripts/docker_build_ml.sh pytorch
`
and get the following when it gets to torchvision:

`Step 15/22 : RUN git clone -b ${TORCHVISION_VERSION} https://github.com/pytorch/vision torchvision &&     cd torchvision &&     python3 setup.py install &&     cd ../ &&     rm -rf torchvision &&     pip3 install "${PILLOW_VERSION}"
 ---> Running in 18d6e6d7c9d0
Cloning into 'torchvision'...
Note: checking out '78ed10cc51067f1a6bac9352831ef37a3f842784'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Traceback (most recent call last):
  File "setup.py", line 13, in <module>
    import torch
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 188, in <module>
    _load_global_deps()
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 141, in _load_global_deps
    ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libcurand.so.10: cannot open shared object file: No such file or directory
`

dusty-nv, 2020-11-16:
What version of JetPack-L4T are you running on your Jetson?

You can check by running `cat /etc/nv_tegra_release`.   If you are running L4T version < R32.4.4 but >= R32.4.2, you can check-out one of these branches:

* https://github.com/dusty-nv/jetson-containers/tree/L4T-R32.4.2
* https://github.com/dusty-nv/jetson-containers/tree/L4T-R32.4.3

If your JetPack version is older than JetPack 4.4, you would need to upgrade JetPack versions to build it successfully.

aldrichg9, 2020-11-16:
Thanks Dusty for the reply.
I am running R32.4.4, just downloaded the fresh image last Tuesday.

> What version of JetPack-L4T are you running on your Jetson?
>
> You can check by running `cat /etc/nv_tegra_release`. If you are running L4T version < R32.4.4 but >= R32.4.2, you can check-out one of [...]

dusty-nv, 2020-11-16:
> I am running R32.4.4, just downloaded the fresh image last Tuesday.

Ah right, sorry I forgot to mention that you also need to set the default docker-runtime to `nvidia` - see here:

https://github.com/dusty-nv/jetson-containers#docker-default-runtime

This is so that CUDA is available during build-time (i.e. when building torchvision in the container)

aldrichg9, 2020-11-16:
Ya that would do it.
Thanks for adding.
Currently troubleshooting, as I'm getting an error about what was added.
'
unable to configure the Docker daemon with file /etc/docker/daemon.json: invalid character '"' after object key:value pair
'

> Ah right, sorry I forgot to mention that you also need to set the default docker-runtime to `nvidia` - see here:
>
> https://github.com/dusty-nv/jetson-containers#docker-default-runtime
>
> This is so [...]