dusty-nv/jetson-containers #36: How to run GUI from docker file

kvnptl, 2021-01-09:
How can I open GUI for Rviz, gedit, Gazebo in the docker container?

System:
- Board: Jetson AGX Xavier 
- Jetpack: 4.4.1

dusty-nv, 2021-01-11:
Hi @kvnptl , I'm not sure about those applications in particular, but if you use the [`scripts/docker_run.sh`](https://github.com/dusty-nv/jetson-containers/blob/8432b19be10e0c81ce8bbf9ceff3f332cf318f25/scripts/docker_run.sh#L104) script, it will automatically setup the container / user permissions so that the container can run GUI applications:

https://github.com/dusty-nv/jetson-containers/blob/8432b19be10e0c81ce8bbf9ceff3f332cf318f25/scripts/docker_run.sh#L104

This involves three things:

1.  Running `sudo xhost +si:localuser:root` before starting the container
2.  Having `-e DISPLAY=$DISPLAY` on your `docker run` command-line
3.  Having `-v /tmp/.X11-unix/:/tmp/.X11-unix` on your `docker run` command-line

For example, in the [Hello AI World](https://github.com/dusty-nv/jetson-inference) project, I run OpenGL windows from inside the Docker containers.

aldrichg9, 2021-01-22:
Yus!!!
This stuff helped me with what I have been fighting.

kvnptl, 2021-06-02:
Thanks, it worked fine for Gvim and I can access the GUI.

But with some ROS packages like Rviz and Gazebo, I am getting the below errors. 

```
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libvulkan.so.1.2 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libdrm_nvdc.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libcuda.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libv4l2.so.0.0.999999 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libv4lconvert.so.0.0.999999 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libnvidia-ptxjitcompiler.so.440.18 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libnvidia-eglcore.so.32.5.0 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libnvidia-glsi.so.32.5.0 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libgstnvegl-1.0.so.0 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libgstnvivameta.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libnvidia-tls.so.32.5.0 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libgstnvexifmeta.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libnvsample_cudaprocess.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libnvidia-fatbinaryloader.so.440.18 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libnvidia-glcore.so.32.5.0 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvofsdk.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvdla_compiler.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvjpeg.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvdla_runtime.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libtegrav4l2.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvargus.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvcam_imageencoder.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvtvmr.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvidia-egl-wayland.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libv4l2_nvvideocodec.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libGLX_nvidia.so.0 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvtnr.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvmm_utils.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvmmlite_video.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvfnetstorehdfx.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvidia-egl-wayland.so.1 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvgov_gpucompute.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvidia-ptxjitcompiler.so.440.18 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvcapture.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvbuf_utils.so.1.0.0 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvgbm.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvapputil.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvidia-eglcore.so.32.5.0 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvfnet.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvcamv4l2.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvbufsurftransform.so.1.0.0 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvphsd.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnveglstream_camconsumer.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvddk_2d_v2.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvtracebuf.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvidia-glsi.so.32.5.0 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvrm_gpu.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvomxilclient.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvgov_generic.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvmmlite_utils.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvmmlite.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnvgov_il.so is empty, not checked.
[... cut ...]
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libnveglstreamproducer.so is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra/libvulkan.so.1.2.141 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra-egl/libGLESv1_CM_nvidia.so.1 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra-egl/libEGL_nvidia.so.0 is empty, not checked.
/sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/tegra-egl/libGLESv2_nvidia.so.2 is empty, not checked.

```

closed 2025-02-21 by johnnynunez