dusty-nv/jetson-containers #40: Docker x86

oscarthorn, 2021-01-12:
Hi!

I know this is only tangentially related to this repo, but you seem to know a lot about the topic and I can't seem to get an answer anywhere.

Basically I want to build a container for jetson with the trt_pose repo from nvidia installed. (https://github.com/NVIDIA-AI-IOT/trt_pose) But I want to do the building on a x86 system and run on Xavier NX.

Here is a minimal docker file:

`# This includes L4T (with CUDA etc) and PyTorch 1.6
FROM nvcr.io/nvidia/l4t-pytorch:r32.4.4-pth1.6-py3

# Use bash from here
SHELL ["/bin/bash", "-c"]

# Torch2TRT
RUN cd ~ \
 && git clone https://github.com/NVIDIA-AI-IOT/torch2trt \
 && cd torch2trt && python3 setup.py install --plugins

# Trt_pose
RUN pip3 install tqdm cython pycocotools \
 && apt-get install -y python3-matplotlib \
 && cd ~ \
 && git clone https://github.com/NVIDIA-AI-IOT/trt_pose \
 && cd trt_pose && python3 setup.py install

# Start command line on start
CMD ["/bin/bash"]` 

I have nvidia-docker installed and as default runtime. Qemu support and cuda cross-compile installed.

But I get the following error:

`Cloning into 'torch2trt'...
Traceback (most recent call last):
  File "setup.py", line 2, in <module>
    import torch
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 188, in <module>
    _load_global_deps()
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 141, in _load_global_deps
    ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libcurand.so.10: cannot open shared object file: No such file or directory
`

What can I do to solve it?

dusty-nv, 2021-01-12:
I haven't tried cross-compiling Docker containers on x86 - I think the issue you are running into, is normally the CUDA toolkit is mapped into the container on Jetson by the nvidia runtime.   However since you are on qemu without JetPack, those files are not there.  

You may be able to cross-compile with CUDA like done here:   https://github.com/NVIDIA/nvidia-docker/wiki/NVIDIA-Container-Runtime-on-Jetson#building-jetson-containers-on-an-x86-workstation-using-qemu

However using additional libraries that load torch and need runtime CUDA components I'm not sure will work, since you are not in an actual JetPack environment on x86.   I would just build it on Xavier NX - I use that device for building all my containers.

oscarthorn, 2021-01-13:
Thanks for the response! That was my worry but I hoped there was a solution. Building on the NX works but it makes setting up a CI/CD pipeline a bit more complicated.

closed 2021-01-13 by oscarthorn

znmeb, 2021-01-13:
> 
> 
> Thanks for the response! That was my worry but I hoped there was a solution. Building on the NX works but it makes setting up a CI/CD pipeline a bit more complicated.

I looked briefly at setting up Travis CI to build the non-GPU Docker images of my project (https://github.com/edgyR/edgyR-containers) on both X86 via "buildx - qemu" and on AWS Gravitons. Unfortunately, neither can complete the longest step of the process - a GHC compile of `pandoc` - fast enough to avoid getting killed. And Travis has shut down their free minutes for open source projects, so that step of my process is currently stuck on my AGX Xavier.

I *think* there's a way to wire up GitLab CI/CD to a machine in your local area network but that's a lot of setup, including moving everything to GitLab, and I already have a `bash` script that does all the building and pushing.

otamajakusi, 2021-10-30:
@oscarthorn @dusty-nv 
This should help for this issue!
https://medium.com/@hiroyuki.obinata/3-times-faster-than-native-build-nvidia-jetson-12fbb3d30450