dusty-nv/jetson-containers #41: ROS foxy build fails

maxtestsymphony, 2021-01-28:
Hi. When I'm trying to build ROS Foxy Dockerfile it is keeps on failing on step:
`RUN cd ${ROS_ROOT} && colcon build --symlink-install`

The error code is:

> Starting >>> rosidl_typesupport_introspection_cpp
--- stderr: libyaml_vendor
CMake Error at CMakeLists.txt:108 (target_link_directories):
  Unknown CMake command "target_link_directories".
  
---
Failed   <<< libyaml_vendor [6.84s, exited with code 1]
Aborted  <<< rosidl_typesupport_introspection_cpp [5.98s]
Aborted  <<< rmw [1min 37s]
Aborted  <<< fastrtps [14min 48s]

Summary: 88 packages finished [17min 19s]
  1 package failed: libyaml_vendor
  3 packages aborted: fastrtps rmw rosidl_typesupport_introspection_cpp
  5 packages had stderr output: cyclonedds foonathan_memory_vendor libyaml_vendor mimick_vendor rcutils
  108 packages not processed
The command '/bin/sh -c cd ${ROS_ROOT} && colcon build --symlink-install' returned a non-zero code: 1

Can you please advise me with this?
P.S: I was trying to remove cmake version 3.10.2 which is installing by default and install 3.13.5 but it doesn't really helped me

bigrobinson, 2021-02-07:
I'm having the same issue. Funny thing is, the last commit to that CMakeLists.txt in libyaml_vendor was apparently to fix this very problem: [(https://github.com/ros2/libyaml_vendor/commit/dfa05c83037fb501948f248aad336c64bcc8ce49)]. And that was back on December 14. I built this ros2 foxy image just a couple of weeks ago with no hitches.

bigrobinson, 2021-02-07:
Here's your hacky workaround until things are straightened out. After the last step of importing the ROS2 source code (currently on line 83 in Dockerfile.ros.foxy), get the latest commit of the libyaml_vendor CMakeLists.txt and overwrite the older commit in the ROS2 source code:

```# Put correct CMakeLists.txt file in libyaml_vendor to avoid cmake error
RUN rm ${ROS_ROOT}/src/libyaml_vendor/CMakeLists.txt && \
    wget --no-check-certificate https://raw.githubusercontent.com/ros2/libyaml_vendor/master/CMakeLists.txt -P ${ROS_ROOT}/src/libyaml_vendor/
```

maxtestsymphony, 2021-02-08:
Thank you a lot for your time @bigrobinson . It helped me and now working)

dusty-nv, 2021-03-31:
It seems this issue still persists and the ROS2 Foxy sources haven't been patched, so I've committed @bigrobinson workaround to master in https://github.com/dusty-nv/jetson-containers/commit/1e10908a104494a883f6855d1e9947827f2a17bc .  Thanks!

closed 2025-02-21 by johnnynunez