dusty-nv/jetson-containers #43: Using docker  command without sudo

mzahana, 2021-01-31:
Greetings,

I would like to use the `docker `command without `sudo` on Jetson Nano, for example `docker run ...` & `docker build...` etc.

I followed the instructions mentioned [here](https://docs.docker.com/engine/install/linux-postinstall/#manage-docker-as-a-non-root-user), but still  I get the following error
```bash
ERRO[0000] failed to dial gRPC: cannot connect to the Docker daemon. Is 'docker daemon' running on this host?: dial unix /var/run/docker.sock: connect: permission denied 
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.40/build?buildargs=%7B%22BASE_IMAGE%22%3A%22nvcr.io%2Fnvidia%2Fl4t-base%3Ar32.4.4%22%7D&cachefrom=%5B%5D&cgroupparent=&cpuperiod=0&cpuquota=0&cpusetcpus=&cpusetmems=&cpushares=0&dockerfile=Dockerfile.ros.melodic.px4&labels=%7B%7D&memory=0&memswap=0&networkmode=default&rm=1&session=x10bzcexctbwk772bu1welkbp&shmsize=0&t=ros%3Amelodic-ros-px4-l4t-r32.4.4&target=&ulimits=null&version=1: dial unix /var/run/docker.sock: connect: permission denied

```
How can I solve this problem?

Thanks

mockbot, 2021-01-31:
add your username to docker group at /etc/group and relogin
e.g.
docker:x:127:nano

mzahana, 2021-02-01:
@mockbot Thanks for your reply. I check the `/etc/group` file and the `docker:x:127username` is already there, but I am still get the following error when I execute `docker ps -a`
```bash

Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Get http://%2Fvar%2Frun%2Fdocker.sock/v1.40/containers/json?all=1: dial unix /var/run/docker.sock: connect: permission denied
```

mzahana, 2021-02-01:
@mockbot it actually worked!
I am not sure adding the user to the `dialout` group helped in that or not, but it works now. Closing the issue.

closed 2021-02-01 by mzahana

Harish1810, 2022-01-10:
@mzahana After adding the username to /etc/group, restarting the Jetson worked for me.