dusty-nv/jetson-containers #44: A custom container built with CUDA dies on start

sskorol, 2021-02-05:
Hi,

I've tried to build a custom docker image based on `nvcr.io/nvidia/l4t-ml:r32.5.0-py3` and `nvcr.io/nvidia/l4t-base:r32.5.0` with CUDA support on Jetson Xavier NX. The build itself has successfully finished, but when I try to run my container either with docker or docker-compose, it fails to start and immediately exits with 132 code. Don't see anything specific in logs:

```shell script
container die 2846517886ea47c56e93c36ddf61a57f7728ea4e93a88b2f349f140e0b280c44 (exitCode=132, image=sskorol/ws, name=bold_gagarin)
```

Here's the Dockerfile:
```docker
FROM nvcr.io/nvidia/l4t-base:r32.5.0

RUN apt-get update && apt-get install -y --no-install-recommends \
	make \
	cmake \
	g++ \
	git \
	swig \
	nano \
	python3 \
	python3-pip \
	python3-dev \
	gcc-8 \
	g++-8 \
	gfortran \
	gfortran-8 \
	automake \
	autoconf \
	unzip \
	libtool \
	subversion

RUN update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-8 8 --slave /usr/bin/g++ g++ /usr/bin/g++-8 --slave /usr/bin/gcov gcov /usr/bin/gcov-8
RUN update-alternatives --install /usr/bin/gfortran gfortran /usr/bin/gfortran-8 8

RUN pip3 install setuptools wheel cython

RUN echo "Preparing Kaldi..." && git clone -b lookahead-1.8.0 --single-branch https://github.com/alphacep/kaldi /opt/kaldi
RUN cd /opt/kaldi/tools && \
	sed -i 's:status=0:exit 0:g' extras/check_dependencies.sh && \
	sed -i 's:CXXFLAGS = -g -O3 -msse -msse2:CXXFLAGS = -g -O3 -march=armv8.2-a -mcpu=cortex-a76:g' Makefile && \
	sed -i 's:--enable-ngram-fsts:--enable-ngram-fsts --disable-bin:g' Makefile && \
	echo "Making openfst..." && \
	make -j $(nproc) openfst cub
RUN cd /opt/kaldi/tools && echo "Making openblas..." && extras/install_openblas_clapack.sh

RUN cd /opt/kaldi/src && \
	./configure --mathlib=OPENBLAS_CLAPACK --shared && \
	sed -i 's: -O1 : -O3 -march=armv8.2-a :g' kaldi.mk && \
	echo "Making kaldi..." && \
	make -j $(nproc) online2 lm rnnlm

RUN echo "Making vosk..." && git clone -b gpu --single-branch https://github.com/sskorol/vosk-api /opt/vosk-api
RUN cd /opt/vosk-api/python && \
	KALDI_MKL=0 KALDI_ROOT=/opt/kaldi KALDI_CUDA=1 python3 ./setup.py install --single-version-externally-managed --root=/

RUN echo "Cleaning up..." && rm -rf /opt/vosk-api && \
	rm -rf /opt/kaldi && \
	rm -rf /root/.cache && \
	rm -rf /var/lib/apt/lists/*
```

When this image is built, I use it as a base for my python code. Nothing specific:
```docker
FROM sskorol/vosk-api

WORKDIR app
COPY . /app

RUN pip3 install -r requirements.txt

EXPOSE 2701

CMD ["python3", "main.py"]
```

I tried to run it the follogin way:
```shell script
docker run --gpus all --net=host --rm --runtime nvidia -e VOSK_LANG=ru -e VOSK_SAMPLE_RATE=16000 -e VOSK_HOST=0.0.0.0 -e VOSK_PORT=2701 -e PYTHONUNBUFFERED=1 -v $PWD/model-ru:/app/model-ru sskorol/ws
```

As well as with docker-compose:
```yaml
version: '3.7'

services:
  ws:
    image: "sskorol/ws:latest"
    ports:
      - "2701:2701"
    environment:
      - VOSK_LANG=ru
      - VOSK_SAMPLE_RATE=16000
      - VOSK_HOST=0.0.0.0
      - VOSK_PORT=2701
    volumes:
      - "${PWD}/model-ru:/app/model-ru"
    networks:
      - vosk
    deploy:
      resources:
        reservations:
          devices:
            - capabilities:
              - gpu
networks:
  vosk:
    driver: bridge
```

Note that I also enabled default nvidia runtime in daemon config:
```json
{
    "data-root": "/home/[USERNAME]/nvme/docker",
    "runtimes": {
        "nvidia": {
            "path": "nvidia-container-runtime",
            "runtimeArgs": []
        }
    },
    "default-runtime": "nvidia"
}
```

Docker version:
```shell script
Client:
 Version:           19.03.6
 API version:       1.40
 Go version:        go1.12.17
 Git commit:        369ce74a3c
 Built:             Fri Dec 18 12:25:49 2020
 OS/Arch:           linux/arm64
 Experimental:      false

Server:
 Engine:
  Version:          19.03.6
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.12.17
  Git commit:       369ce74a3c
  Built:            Thu Dec 10 13:23:49 2020
  OS/Arch:          linux/arm64
  Experimental:     false
 containerd:
  Version:          1.3.3-0ubuntu1~18.04.4
  GitCommit:
 nvidia:
  Version:          spec: 1.0.1-dev
  GitCommit:
 docker-init:
  Version:          0.18.0
  GitCommit:
```

Would be greatly appreciated any help.

dusty-nv, 2021-02-05:
If you remove the CMD from your dockerfile, does it start?  Are you able to start just a very simple dockerfile that you build yourself?

sskorol, 2021-02-05:
@dusty-nv yes, when I remove CMD, I can run a container in an interactive mode:
```bash script
docker run -it sskorol/ws /bin/bash
```
But I need to run my python server in a daemon mode.

dusty-nv, 2021-02-05:
There is some exception then trying to run your python script.   What happens if you try running your python script interactively?   Like:

```
docker run --runtime nvidia -it sskorol/ws python3 main.py
```

If you are basing your image off l4t-base, use `--runtime nvidia` too

sskorol, 2021-02-05:
@dusty-nv I tried to run it from bash and you were right regarding the exception, but it's not really obvious:
```
Illegal instruction (core dumped)
```

sskorol, 2021-02-05:
Also, note that this python script uses a library that was built in the initial image. Generally speaking, it's just a python wrapper for the C++ code of the ASR engine. In the base image, I've built Kaldi and Vosk ASR with CUDA. I can successfully build and run it on the Jetson Xavier NX host. But the same code is failing in docker.

dusty-nv, 2021-02-05:
I haven't used those specific libraries before and haven't had to explicitly set those GCC flags before, but I recommend putting some print() statements before your imports to see which import is causing the segfault.

sskorol, 2021-02-05:
As expected, it fails on Vosk import, which was initially built in the base image. Ideally, it'd be nice to get a gdb stack, but this package is missing and can't be installed via the package manager.

dusty-nv, 2021-02-05:
Are you able to import vosk from the base image, if you just run the base image and run a python3 interpreter in interactive mode?

sskorol, 2021-02-05:
Not yet, but I'm pretty sure the result will be similar, as there's nothing specific added to the child container. In the meantime, trying to build gdb in a container and then will run a python script in debug mode. Hopefully, it'll give a good low-level stacktrace.

sskorol, 2021-02-05:
gdb gives the following bt:
```
#0  0x0000007f982adbd4 in gotoblas_dynamic_init () from /usr/local/lib/python3.6/dist-packages/vosk/_vosk.cpython-36m-aarch64-linux-gnu.so
#1  0x0000007f97e10e54 in gotoblas_init () from /usr/local/lib/python3.6/dist-packages/vosk/_vosk.cpython-36m-aarch64-linux-gnu.so
#2  0x0000007f9a2bea34 in ?? () from /lib/ld-linux-aarch64.so.1
#3  0x000000001d486110 in ?? ()
```

sskorol, 2021-02-05:
Talked to Kaldi maintainer, he suggested trying to rebuild Openblas w/o DYNAMIC_ARCH flag. As it seems like that stack refers to platform-specific openblas issue. Rebuilding everything and check if it works.

sskorol, 2021-02-06:
Ok, so the problem was in Kaldi scripts which were not adopted for running from within docker on Jetson boards.

I patched https://github.com/alphacep/kaldi/blob/lookahead-1.8.0/tools/extras/install_openblas_clapack.sh#L9 during the build time:
```shell script
sed -i 's:DYNAMIC_ARCH=1 TARGET=NEHALEM:TARGET=ARMV8:g' extras/install_openblas_clapack.sh
```
Openblas was successfully built with Kaldi and Vosk and I was able to run the ASR server with GPU support inside the docker container based on the official [ml](https://ngc.nvidia.com/catalog/containers/nvidia:l4t-ml) image.

@dusty-nv thanks for your tips which helped me a lot to find the root cause!

closed 2021-02-06 by sskorol

dusty-nv, 2021-02-08:
OK gotcha, glad you got it running!