dusty-nv/jetson-containers #46: Install moveit within Dockerfile.ros.noetic

eliabruni, 2021-02-10:
Hi!

I am trying to building Moveit! from source as explained [here](https://moveit.ros.org/install/source/) within `Dockerfile.ros.noetic`, but I am encountering some issues.

Here is what I am trying in the Dockerfile, right after [line 54](https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ros.noetic#L54) of `Dockerfile.ros.noetic`:

```
# Install Moveit dependencies
RUN apt-get update && \
    apt-get install --no-install-recommends --no-install-suggests --yes \
    clang-format-10 \
    && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

# Install Moveit
WORKDIR /
RUN chmod +x ${ROS_ROOT}/setup.bash
RUN /bin/bash -c  '${ROS_ROOT}/setup.bash'

RUN mkdir moveit_catkin_ws && \
    cd moveit_catkin_ws && \
    wstool init src && \
    wstool merge -t src https://raw.githubusercontent.com/ros-planning/moveit/master/moveit.rosinstall && \
    wstool update -t src && \
    # ISSUE HERE 
    rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO} --os=ubuntu:bionic

RUN catkin config --extend /opt/ros/${ROS_DISTRO} --cmake-args -DCMAKE_BUILD_TYPE=Release
```

And this is what I am getting:

```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit_ros_move_group: No definition of [rostest] for OS version [bionic]
moveit_chomp_optimizer_adapter: No definition of [pluginlib] for OS version [bionic]
moveit_resources_prbt_pg70_support: No definition of [xacro] for OS version [bionic]
moveit_planners: No definition of [catkin] for OS version [bionic]
moveit: No definition of [catkin] for OS version [bionic]
moveit_ros_occupancy_map_monitor: No definition of [rosunit] for OS version [bionic]
moveit_ros_planning: No definition of [tf2_ros] for OS version [bionic]
moveit_runtime: No definition of [catkin] for OS version [bionic]
moveit_ros_visualization: No definition of [rostest] for OS version [bionic]
moveit_resources_pr2_description: No definition of [catkin] for OS version [bionic]
moveit_ros_control_interface: No definition of [trajectory_msgs] for OS version [bionic]
moveit_ros_manipulation: No definition of [pluginlib] for OS version [bionic]
moveit_ros_perception: No definition of [rosunit] for OS version [bionic]
moveit_ros_warehouse: No definition of [tf2_ros] for OS version [bionic]
rviz_visual_tools: No definition of [rosunit] for OS version [bionic]
moveit_ros: No definition of [catkin] for OS version [bionic]
moveit_resources_prbt_moveit_config: No definition of [rviz] for OS version [bionic]
moveit_tutorials: No definition of [rosunit] for OS version [bionic]
pilz_industrial_motion_planner: No definition of [code_coverage] for OS version [bionic]
moveit_planners_chomp: No definition of [rostest] for OS version [bionic]
moveit_resources_panda_moveit_config: No definition of [topic_tools] for OS version [bionic]
moveit_fake_controller_manager: No definition of [roscpp] for OS version [bionic]
moveit_simple_controller_manager: No definition of [actionlib] for OS version [bionic]
moveit_msgs: No definition of [std_msgs] for OS version [bionic]
moveit_ros_robot_interaction: No definition of [rosunit] for OS version [bionic]
moveit_resources_prbt_support: No definition of [code_coverage] for OS version [bionic]
moveit_commander: No definition of [rostest] for OS version [bionic]
moveit_resources_prbt_ikfast_manipulator_plugin: No definition of [tf2_kdl] for OS version [bionic]
moveit_resources_fanuc_description: No definition of [catkin] for OS version [bionic]
pilz_industrial_motion_planner_testutils: No definition of [catkin] for OS version [bionic]
moveit_core: No definition of [rosunit] for OS version [bionic]
moveit_servo: No definition of [rostest] for OS version [bionic]
moveit_ros_planning_interface: No definition of [eigen_conversions] for OS version [bionic]
moveit_ros_benchmarks: No definition of [pluginlib] for OS version [bionic]
moveit_kinematics: No definition of [xmlrpcpp] for OS version [bionic]
moveit_plugins: No definition of [catkin] for OS version [bionic]
moveit_resources: No definition of [robot_state_publisher] for OS version [bionic]
moveit_visual_tools: No definition of [cmake_modules] for OS version [bionic]
panda_moveit_config: No definition of [topic_tools] for OS version [bionic]
moveit_resources_panda_description: No definition of [catkin] for OS version [bionic]
chomp_motion_planner: No definition of [catkin] for OS version [bionic]
geometric_shapes: No definition of [rosunit] for OS version [bionic]
moveit_setup_assistant: No definition of [rosunit] for OS version [bionic]
moveit_planners_ompl: No definition of [eigen_conversions] for OS version [bionic]
moveit_resources_fanuc_moveit_config: No definition of [xacro] for OS version [bionic]
The command '/bin/sh -c mkdir moveit_catkin_ws &&     cd moveit_catkin_ws &&     wstool init src &&     wstool merge -t src https://raw.githubusercontent.com/ros-planning/moveit/master/moveit.rosinstall &&     wstool update -t src &&     rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO} --os=ubuntu:bionic' returned a non-zero code: 1
```

AWilson10, 2021-11-21:
Hi,
I have the same issue as described here. I tried building within a container and outside of the container. Is it that moveit can not be built from source on jetson? Even though the noetic desktop can ?
Are there any more ideas om how to resolve this?

eliabruni, 2021-11-21:
Here is how I solved it:

```
RUN apt-get update && \
    apt-get install --no-install-recommends --no-install-suggests --yes \
    build-essential \
    cmake \
    libbullet-dev \
    python3-colcon-common-extensions \
    python3-flake8 \
    python3-pip \
    python3-pytest-cov \
    python3-rosdep \
    python3-setuptools \
    python3-vcstool \
    wget \
    clang-format-10 \
    && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/*

# install some pip packages needed for testing
RUN python3 -m pip install -U \
    argcomplete \
    flake8-blind-except \
    flake8-builtins \
    flake8-class-newline \
    flake8-comprehensions \
    flake8-deprecated \
    flake8-docstrings \
    flake8-import-order \
    flake8-quotes \
    pytest-repeat \
    pytest-rerunfailures \
    pytest

# create workspace
ENV COLCON_WS=/ws_moveit2
RUN mkdir -p $COLCON_WS/src
WORKDIR $COLCON_WS/src

# get repo 
RUN git clone https://github.com/ros-planning/moveit2.git
RUN git clone https://github.com/ros-planning/moveit2_tutorials.git
RUN vcs import < moveit2/moveit2.repos
RUN vcs import < moveit2_tutorials/moveit2_tutorials.repos

# install 
RUN apt-get update && \
    rosdep install -r --from-paths $COLCON_WS/src --ignore-src --rosdistro foxy -y

WORKDIR $COLCON_WS

# Set the gcc default back to 9.3
RUN update-alternatives --remove-all gcc && \
    update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-9 800 --slave /usr/bin/g++ g++ /usr/bin/g++-9

# build moveit2
# moveit2 requires access to the installation of ROS2, hence we need this hacky implementation in which we source
# the installation
RUN /bin/bash -c "source $ROS_ROOT/setup.bash ; colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release"

# source workspace
RUN echo "source $COLCON_WS/install/setup.bash" >> /root/.bashrc
```

AWilson10, 2021-11-22:
Thanks for that - will give it a try - appreciate it.

Kind Regards,

Anthony

T: +44 (0)1224 929010

M: +44(0)7855 264204

Unit 14 Hareness Trade Park, Hareness Road, Altens, AB12 3LE

[image: DNV GL Logo & UKAS 0013]

www.qltech.net

Quantum Leap Technologies (QL Tech) Limited is a limited company registered
in England. Company Registered No 8921426. Registered office: Seaways,
Restronguet Point, Feock, Truro, Cornwall, TR3 6RL.  Private and
confidential: This e-mail transmission is strictly confidential and
intended solely for the person or organisation to whom it is addressed. If
you are not the intended recipient, you must not copy, disclose, distribute
or take any action in reliance on it. If you have received this e-mail in
error, please notify us and then delete it as soon as possible. The sender
of this email, or QL Tech Ltd., is not responsible for any changes made to
it or any attachments after transmission. It is the responsibility of the
recipient to ensure that the onward transmission, opening or use of this
message and any attachments will not adversely affect their systems or
data. Please carry out virus and other such checks as you consider
appropriate.

AWilson10, 2021-11-22:
Hi
Just realised this is for ros2 foxy. We are wanting noetic.
Thanks

eliabruni, 2021-11-22:
Hey, indeed that's why I removed the comment. I am not sure we managed to make Noetic work in the end.

AWilson10, 2021-11-22:
Ok, thanks.

closed 2025-02-21 by johnnynunez