dusty-nv/jetson-containers #49: How do I run the container inside visual studio code ?

MuhammedRishdhi123, 2021-03-03:


dusty-nv, 2021-03-03:
@MuhammedRishdhi123 I haven't used Visual Studio Code, so perhaps some others who are more familiar may have suggestions.

camelator, 2021-03-05:
Yes,
there are several options:
Working on a remote PC, you can have studio code on the PC, and use remote-ssh from the command palette (maybe a plugin is requested, but google is your friend)

Instead, you can also build a new docker image from the image you are currently using and install code-server, which is a WEB code studio. It means you wan directly work from any Web navigator in your network

Here are part of my dockerfile I am currently using:

# code studio

#install dependencies
RUN apt-get update && \
    apt-get -y upgrade
RUN apt-get -y install nodejs-dev node-gyp libssl1.0-dev
RUN apt-get -y install git nano mc curl wget npm make autoconf g++ flex bison libfl2 libfl-dev 

RUN curl -o /tmp/code-server.deb -L https://github.com/cdr/code-server/releases/download/v${CODE_STUDIO_VERSION}/code-server_${CODE_STUDIO_VERSION}_arm64.deb && \
    dpkg -i /tmp/code-server.deb && \
    rm /tmp/code-server.deb
RUN npm install -g configurable-http-proxy
RUN code-server --install-extension ms-python.python && \
    code-server --install-extension eamodio.gitlens && \
    code-server --install-extension ms-vscode.cpptools && \
    code-server --install-extension vscode-icons-team.vscode-icons

closed 2021-03-05 by MuhammedRishdhi123