dusty-nv/jetson-containers #50: Unable to install ros package with noetic container

camelator, 2021-03-05:
Within the noetic container, trying to do:
apt update
apt install ros-noetic-rgbd-launch

return the following error:
 E: Unable to locate package ros-noetic-rgbd-launch

as I can find this package on my PC, I suspect an architecture issue?

dusty-nv, 2021-03-05:
Hi @camelator, ROS Noetic is built from source in the container, because there aren't Noetic packages in apt for Ubuntu 18.04.  So you may need to build this package from source.

camelator, 2021-03-05:
this is what I presumed.
now I am tring to build the package for about 6 hours because of dependencies
It is a shame that Jetson has no official image for Ubuntu 20 :(

MrOCW, 2022-05-22:
hi @dusty-nv in this case, how would you recommend I manage a workspace for large projects with many dependencies? Do I build the dependencies from source with the ROS workspace during the docker build process then create an overlay?

dusty-nv, 2022-05-23:
> Do I build the dependencies from source with the ROS workspace during the docker build process then create an overlay?

Yes, here is one such example that builds/installs packages on top of ROS2 container:   https://github.com/dusty-nv/jetbot_ros/blob/master/Dockerfile

camelator, 2022-05-24:
ok, found the issues I had with dependencies.
I tried to add opencv to the dusty-nv images without success.
What is important is to change the docker parameters to run it with nvidia runtime:
/etc/docker/daemon.json
{
    "runtimes": {
        "nvidia": {
            "path": "/usr/bin/nvidia-container-runtime",
            "runtimeArgs": []
        }
    },

"default-runtime": "nvidia"
 }

Then no issue to integrate cuda libs during a build step

closed 2025-02-21 by johnnynunez