dusty-nv/jetson-containers #52: exec error when using docker file

AWilson10, 2021-03-08:
Hi, I tried to run the build script to build noetic (although it fails at the same place on any script).
I get this error:
standard_init_linux.go:211: exec user process caused "exec format error"
The command '/bin/sh -c apt-get update &&     apt-get install -y --no-install-recommends           git 		cmake 		build-essential 		curl 		wget 		gnupg2 		lsb-release     && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 1

Can you explain why?
I tried running with sudo but no difference
Thanks

dusty-nv, 2021-03-08:
> I am building on a x86 amd machine, could this be the issue?
Hi @AWilson10, yes these containers are meant to be built on Jetson.

AWilson10, 2021-03-12:
Hi. I used this guide to emulate arm and it installed and run up ok on my x86 pc.
https://www.stereolabs.com/docs/docker/building-arm-container-on-x86/

closed 2021-03-12 by AWilson10