dusty-nv/jetson-containers #57: If some modules such as h5py is needed how can i add it to the container

MuhammedRishdhi123, 2021-03-14:


dusty-nv, 2021-03-15:
After you start the container, you will have a terminal prompt.  You can install the packages through the terminal, and then use [`docker commit`](https://docs.docker.com/engine/reference/commandline/commit/) to save your changes to a new container tag.

Alternatively, you can make your own Dockerfile where you use my container as your base image (using the `FROM` statement)

MuhammedRishdhi123, 2021-03-18:
> After you start the container, you will have a terminal prompt. You can install the packages through the terminal, and then use [`docker [...]

Thanks @dusty-nv  I was having trouble with that now it works fine !

closed 2021-03-18 by MuhammedRishdhi123