dusty-nv/jetson-containers #60: JetPack 4.5.1 and Ros2 Foxy (Dockerfile) Installation

the-black-viper, 2021-03-31:
I am trying to install the dockerfile on a freshly flashed image of JP 4.5.1. Here is the snippet of the log after installation:

```
--- stderr: cyclonedds
You have called ADD_LIBRARY for library ddsc without any source files. This typically indicates a problem with your CMakeLists.txt file
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.google.inject.internal.cglib.core.$ReflectUtils$1 (file:/usr/share/maven/lib/guice.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
WARNING: Please consider reporting this to the maintainers of com.google.inject.internal.cglib.core.$ReflectUtils$1
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
---
...
--- stderr: libyaml_vendor
CMake Error at CMakeLists.txt:108 (target_link_directories):
  Unknown CMake command "target_link_directories".

---
Failed   <<< libyaml_vendor [7.83s, exited with code 1]
...
Summary: 88 packages finished [20min 22s]
  1 package failed: libyaml_vendor
  3 packages aborted: fastrtps rmw rosidl_typesupport_introspection_cpp
  5 packages had stderr output: cyclonedds foonathan_memory_vendor libyaml_vendor mimick_vendor rcutils
  108 packages not processed

```
I can install eloquent just fine however I am having trouble with foxy.

dusty-nv, 2021-03-31:
Hi @the-black-viper, can you try pulling the latest updates from this repo and try building again?

I just committed [this workaround](https://github.com/dusty-nv/jetson-containers/issues/41#issuecomment-774767272) for the issue in https://github.com/dusty-nv/jetson-containers/commit/1e10908a104494a883f6855d1e9947827f2a17bc

the-black-viper, 2021-04-05:
@dusty-nv I just built the image without issues using the latest updates. Closing this thread.

closed 2021-04-05 by the-black-viper