dusty-nv/jetson-containers #61: looking to use nano GPIO from the docker

wegunterjrFIrefly, 2021-04-06:
From what I gather, this might be a two part solution.  
First, solve how to connect or access the GPIO generally
Second, how to expose them to the container. 

Thoughts?

dusty-nv, 2021-04-06:
Here are a couple forum topics I found about GPIO inside container:

https://forums.developer.nvidia.com/t/use-gpio-from-l4t-ml-container/129220/3?u=dusty_nv
https://forums.developer.nvidia.com/t/using-gpio-from-within-a-docker-container/110260/15?u=dusty_nv

If you run the container with `-v /sys:/sys` can you see the GPIO device files like these?

```
$ echo 220 > /sys/class/gpio/export 
$ echo out > /sys/class/gpio/gpio220/direction
$ echo 1 > /sys/class/gpio/gpio220/value
```

wegunterjrFIrefly, 2021-04-06:
`root@robot# echo 220 > /sys/class/gpio/export
bash: echo: write error: Device or resource busy
root@robot# echo out > /sys/class/gpio/gpio220/direction
root@robot# echo 1 > /sys/class/gpio/gpio220/value
`

dusty-nv, 2021-04-06:
It seems you can see the /sys GPIO files.  Here is a description of what they do:   https://www.kernel.org/doc/Documentation/gpio/sysfs.txt

wegunterjrFIrefly, 2021-04-07:
ah...had to do these steps to make sure I had permissions, then restarted the rules:
```
service udev restart 
udevadm control -R 
udevadm trigger 
```

Now, to figure out the right mapping or use RPi.GPIO or the Jetson.GPIO?   
```
gpio-0   (                    )
 gpio-1   (                    )
 gpio-2   (                    |pcie_wake           ) in  hi    
 gpio-3   (                    )
 gpio-4   (                    )
 gpio-5   (                    )
 gpio-6   (                    |system-suspend-gpio ) out hi    
 gpio-7   (                    )
 gpio-8   (                    )
 gpio-9   (                    )
 gpio-10  (                    )
 gpio-11  (                    )
 gpio-12  (SPI1_MOSI           |sysfs               ) out lo    
 gpio-13  (SPI1_MISO           )
 gpio-14  (SPI1_SCK            )
 gpio-15  (SPI1_CS0            )
 gpio-16  (SPI0_MOSI           )
 gpio-17  (SPI0_MISO           )
 gpio-18  (SPI0_SCK            )
 gpio-19  (SPI0_CS0            )
 gpio-20  (SPI0_CS1            )
 gpio-21  (                    )
 gpio-22  (                    )
 gpio-23  (                    )
 gpio-24  (                    )
 gpio-25  (                    )
 gpio-26  (                    )
 gpio-27  (                    )
 gpio-28  (                    )
 gpio-29  (                    )
 gpio-30  (                    )
 gpio-31  (                    )
 gpio-32  (                    )
 gpio-33  (                    )
 gpio-34  (                    )
 gpio-35  (                    )
 gpio-36  (                    )
 gpio-37  (                    )
 gpio-38  (GPIO13              )
 gpio-39  (                    )
 gpio-40  (                    )
 gpio-41  (                    )
 gpio-42  (                    )
 gpio-43  (                    )
 gpio-44  (                    )
 gpio-45  (                    )
 gpio-46  (                    )
 gpio-47  (                    )
 gpio-48  (                    )
 gpio-49  (                    )
 gpio-50  (UART1_RTS           )
 gpio-51  (UART1_CTS           |sysfs               ) out hi    
 gpio-52  (                    )
 gpio-53  (                    )
 gpio-54  (                    )
 gpio-55  (                    )
 gpio-56  (                    |sysfs               ) out lo    
 gpio-57  (                    )
 gpio-58  (                    )
 gpio-59  (                    )
 gpio-60  (                    )
[... 155 lines cut ...]
 gpio-216 (GPIO09              )
 gpio-217 (                    )
 gpio-218 (                    )
 gpio-219 (                    )
 gpio-220 (                    |sysfs               ) out hi    
 gpio-221 (                    )
 gpio-222 (                    )
 gpio-223 (                    )
 gpio-224 (                    )
 gpio-225 (                    |hdmi2.0_hpd         ) in  lo IRQ
 gpio-226 (                    )
 gpio-227 (                    )
 gpio-228 (                    |extcon:extcon@1     ) in  hi IRQ
 gpio-229 (                    )
 gpio-230 (                    )
 gpio-231 (                    |?                   ) out hi    
 gpio-232 (SPI1_CS1            )
 gpio-233 (                    )
 gpio-234 (                    )
 gpio-235 (                    )
 gpio-236 (                    )
 gpio-237 (                    )
 gpio-238 (                    )
 gpio-239 (                    )
```

wegunterjrFIrefly, 2021-04-07:
Got it figured out, I think .  The mapping is sort of handled inside the Jetson.GPIO/RPi.GPIO

I did have to setup the pins function to enable the PWM.   
`sudo /opt/nvidia/jetson-io/jetson-io.py`

Then just had to trust it.  

Also, don't forget to cleanup:
`   GPIO.cleanup()`

closed 2021-04-07 by wegunterjrFIrefly