dusty-nv/jetson-containers #64: ROS Noetic is not create the /opt/ros folder

sinasadeghian, 2021-05-11:
Hi,
I just installed the ros noetic, however, I don't the /opt/ros to setup it in bashrc. What should I do?

dusty-nv, 2021-05-11:
> however, when I do the ls, I don't see the ros folder to set the source in bashrc
> 
> /opt$ ls
> containerd nvidia ota_package

You aren't running in the container when you do that.  To start the container, run `./scripts/docker_run.sh -c ros:noetic-ros-base-l4t-r32.4.4`

It will give you a terminal inside the container - then you should find it under `/opt`
BTW, the `/opt/ros/noetic/setup.bash` is automatically sourced when the container starts.

sinasadeghian, 2021-05-11:
:~/jetson-containers$ ./scripts/docker_run.sh -c ros:noetic-ros-base-l4t-r32.4.4
localuser:root being added to access control list
bash: /opt/ros/noetic/share/ros/setup.bash: No such file or directory

I'm getting this error. still it is not adding to the opt folder. 

However, in the super user mode I could see the opt folder in root and ros in it.
when I do cd /opt ros is not there.  the container is adding ros to opt folder not /opt
How could I fix this?

dusty-nv, 2021-05-11:
What I don't understand is how `scripts/docker_test_ros.sh noetic` works ok if it is not in fact there.

Where did you find ROS at inside the container?

You can also try this container image instead - https://github.com/NVIDIA-AI-IOT/ros2_jetson/tree/main/docker#pull-docker-images-from-docker-hub-follow

sinasadeghian, 2021-05-11:
I tried the image container that you recommended.
It is not installing completely. It is showing this error
Step 19/45 : RUN apt-get update &&     apt-get install -y vim &&     apt-get install -y rviz
 ---> Using cache
 ---> e220c448bc06
Step 20/45 : COPY jetson-ota-public.asc /etc/apt/trusted.gpg.d/jetson-ota-public.asc
COPY failed: stat /var/lib/docker/tmp/docker-builder476182848/jetson-ota-public.asc: no such file or directory

What should I do?

dusty-nv, 2021-05-12:
> Step 20/45 : COPY jetson-ota-public.asc /etc/apt/trusted.gpg.d/jetson-ota-public.asc
> COPY failed: stat /var/lib/docker/tmp/docker-builder476182848/jetson-ota-public.asc: no such file or directory

You can find this file on your device at the same location - `/etc/apt/trusted.gpg.d/jetson-ota-public.asc`

Copy it into the same directory on your device that you are building the Dockerfile from.

AmirsamanAhmadi, 2021-05-27:
@dusty-nv I also have same problem I successfully managed to install the neotic however i do not know how to run this container and add my workspace into the .bashrc file. where this ros is stored ?
I used the provided command `./scripts/docker_run.sh -c ros:noetic-ros-base-l4t-r32.4.4`  and i am getting exact same error as @sinasadeghian.

dusty-nv, 2021-05-27:
@AmirsamanAhmadi @sinasadeghian after debugging this, I found that `$ROS_ROOT` gets set by noetic to `/opt/ros/noetic/share/ros`, which was messing up sourcing the environment in `/root/.bashrc`.    So I fixed this in https://github.com/dusty-nv/jetson-containers/commit/857b413a4dba304a06754df8a03c76119ada2138

ROS Noetic is stored inside the container under `/opt/ros/noetic`
The .bashrc is found at `/root/.bashrc`

I also uploaded a pre-built container image of this to [`dustynv/ros:noetic-ros-base-l4t-r32.5.0`](https://hub.docker.com/r/dustynv/ros/tags?page=1&ordering=last_updated)

closed 2025-02-21 by johnnynunez