dusty-nv/jetson-containers #65: l4t-ros-noetic-pytorch:r32.5 contains melodic package, and OSRF ros images

pvmilk, 2021-05-26:
I tried to use nvidiajetson/l4t-ros-noetic-pytorch:r32.5 by pulling from dockerhub using
```
$ docker pull nvidiajetson/l4t-ros-noetic-pytorch:r32.5
```

Then, I try to install noetic package inside, but can only find package that started with 'ros-melodic-PACKAGE'.

I decided to use the ros package from osrf instead, i.e.
```
$ docker pull arm64v8/ros:melodic-perception
```
Would like to know that apart from those ML packages, are there any different between 
1.) `l4t-ros-XXX' provided by nvidia 
2.) `arm64v8/ros:XXX` provided by osrf

pvmilk, 2021-05-26:
I think I found my own answer on the second issues:

https://gitlab.com/nvidia/container-images/l4t-base/-/tree/master

dusty-nv, 2021-05-26:
> https://gitlab.com/nvidia/container-images/l4t-base/-/tree/master

Use `nvcr.io/nvidia/l4t-base:r32.4.4` instead of that l4t-base you linked to.  The gitlab doesn't contain the same way the actual l4t-base was built.  It is a special internal build process for l4t-base since CUDA/cuDNN/TensorRT are mounted into the container dynamically at runtime.

For your issue with `nvidiajetson/l4t-ros-noetic-pytorch:r32.5`, please log this bug on https://github.com/NVIDIA-AI-IOT/ros2_jetson so my colleagues who made these other container images can take a look.  Thanks.

closed 2025-02-21 by johnnynunez