dusty-nv/jetson-containers #66: Using opencv VideoCapture

Noam-M, 2021-05-28:
Hello @dusty-nv,

Iv'e been using the l4t-ml docker container using images saved locally on my Jetson Xavier NX, and now I'm interested in using my network directly using my USB webcam.

I have successfully connected it and I see it as ` '/dev/video0'`.
I'm using the following code but cannot capture the actual images:
```
cap = cv2.VideoCapture(0)
ret, frame = cap.read()
print(frame)
```
Which simply outputs `None`

I have also tried to mount the` '/dev' `directory onto the docker image as follows:
`sudo docker run -it --rm --runtime nvidia --network host -v /dev:/dev nvcr.io/nvidia/l4t-ml:r32.4.4-py3`
I can see `'/dev/video0` inside the container but unfortunately it didn't help using VideoCapture

Do you have an idea what's missing?

Thanks!

dusty-nv, 2021-05-28:
Hmm.  If you run this inside container, can it print information about your /dev/video0 device?

``` bash
apt-get update && apt-get install v4l-utils
v4l2-ctl --device /dev/video0 --list-formats-ext
```

Noam-M, 2021-05-29:
Hello @dusty-nv 

Thank you for the quick reply!
I have tried that in my container and got the following output:
`Failed to open /dev/video0: Operation not permitted`
Doing the same thing outside the container worked flawlessly
How can I change the permissions for the operation?

Thanks

dusty-nv, 2021-05-29:
Hmm. Can you try mounting /dev/video0 instead of /dev

Noam-M, 2021-05-31:
Hello @dusty-nv 

I tried this:
`sudo docker run -it --rm --runtime nvidia --network host -v /dev/video0:/dev/video0 nvcr.io/nvidia/l4t-ml:r32.4.4-py3`
And still the VideoCapture didn't work (got `None`)
Trying to run `v4l2-ctl --device /dev/video0 --list-formats-ext` from within the container,
I got `Failed to open /dev/video0: Is a directory` as output

I then tried to run the docker as follows: `sudo docker run -it --rm --runtime nvidia --network host -v /dev/video0:/dev nvcr.io/nvidia/l4t-ml:r32.4.4-py3`
But it didn't run at all, issuing the following error:
`docker: Error response from daemon: OCI runtime create failed: container_linux.go:367: starting container process caused: process_linux.go:495: container init caused: open /dev/ptmx: no such file or directory: unknown.
`

Any further ideas?

Thanks!

Noam-M, 2021-06-06:
Hello @dusty-nv 
Any other ideas of how to use a video from a webcam in real time?

Thanks!

dusty-nv, 2021-06-07:
Hmm...what is the USB webcam you are using?

Noam-M, 2021-06-11:
Hello @dusty-nv 

My webcam is Xiaomi IMILAB Full HD 1080P Webcam W88 S.

Noam-M, 2021-06-20:
Hello @dusty-nv 
Any suggestions?

Thanks!

dusty-nv, 2021-06-21:
Hi @Noam-M, sorry for the delay.  Can you try mounting `/dev/video0` with the `--device` flag instead?

``` bash
sudo docker run -it --rm --runtime nvidia --network host --device /dev/video0 nvcr.io/nvidia/l4t-ml:r32.4.4-py3
```

sudhamsursr, 2022-03-10:
Hi,

The last command posted by @dusty-nv has worked for me on Jetson Xavier:
`sudo docker run -it --rm --runtime nvidia --network host --device /dev/video0 nvcr.io/nvidia/l4t-ml:r32.4.4-py3`

However, if you have multiple cameras, the following command had worked for me: (4 cameras for example)
`sudo docker run -it --rm --runtime nvidia --network host --device /dev/video0 --device /dev/video1 --device /dev/video2 --device /dev/video4 nvcr.io/nvidia/l4t-ml:r32.4.4-py3`

closed 2025-02-21 by johnnynunez