dusty-nv/jetson-containers #69: OSRF GPG Expiry Incident

ARLunan, 2021-06-05:
Are these ROS files and installation scripts still valid under the May 29, 2021 extended Keys expiry date measures?

dusty-nv, 2021-06-07:
Hi @ARLunan, just made the updates for this - see https://github.com/dusty-nv/jetson-containers/pull/67

Tiryoh, 2021-06-07:
I guess merging #67 and building the ROS 2 image would solve this problem for ROS 2, but it is not the root solution to this problem. #67 reflects changes made in April 2021, but this GPG key expiration happened in May 2021.
The scope of influence of the GPG Key Expiration is both ROS 1 and ROS 2, which requires reinstalling or updating the GPG key.
ref: https://discourse.ros.org/t/ros-gpg-key-expiration-incident/20669

I think there are two ways to solve this problem.

1. Include the following command in the README of this repository and ask users to update their GPG keys.
    ```
    apt update || apt install -y curl && curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | apt-key add -
    ```
    ref: https://discourse.ros.org/t/ros-gpg-key-expiration-incident/20669/28
2. Rebuild the ROS images, including ROS 1 Melodic and Noetic.
    * The new GPG key will be imported at rebuild time, so that should solve the problem.

dusty-nv, 2021-06-07:
Thanks for the clarification, @Tiryoh.  I will rebuild the ROS1 Melodic and Noetic containers too and update the images on DockerHub.

dusty-nv, 2021-06-07:
OK, [these ROS containers](https://github.com/dusty-nv/jetson-containers#pre-built-container-images) have been updated on DockerHub.  Thanks everyone!

closed 2025-02-21 by johnnynunez