dusty-nv/jetson-containers #71: Can not connect Raspberry Pi camera (CSI) inside ROS container (Xavier NX)

mzahana, 2021-06-21:
Hello.

I am trying to run a RPi camera, which is connected to a CSI port on Xavier NX, inside the ROS melodic container. When I execute the following command
```
gst-launch-1.0 nvarguscamerasrc ! nvoverlaysink
```

I get the following error
```
ERROR: pipeline could not be constructed: no element "nvarguscamerasrc".
```

However, tt runs fine outside the container.

When I run the following command inside the ROS container,
```
gst-inspect-1.0 -`
```
it shows that the `nvarguscamerasrc` is blacklisted
```
Blacklisted files:
  libgstnvarguscamerasrc.so
  libgstnvdrmvideosink.so
  libgstnvivafilter.so
  libgstnvvideo4linux2.so
  libgstnvvidconv.so
  libgstnvv4l2camerasrc.so
  libgstnvcompositor.so
  libgstnvjpeg.so
  libgstomx.so
  libgstnvvideosinks.so
```

Any idea how  to solve this issue so I can connect tot eh RPi camera inside the container?

Thanks

dusty-nv, 2021-06-21:
Hi @mzahana, can you start your container with `--runtime nvidia` and `-v /tmp/argus_socket:/tmp/argus_socket` ?

mzahana, 2021-06-22:
@dusty-nv Thanks! that worked!
My main problem is that I was creating custom Dockerfiles based on the ones in this repo, but with added user account (not root), and that seems to be causing issues some how. I fell back to root user and it worked, and no more blacklisted libraries by gstreamer.

closed 2021-06-22 by mzahana

dusty-nv, 2021-06-22:
Yea typically a user needs to belong to `VIDEO` group in order to use GPU stuff, however I'm not sure if/how this works within Docker.

mzahana, 2021-06-23:
@dusty-nv I have another issue.
I am able to access RPi camera inside the container, but I am not able to access it in Python using OpenCV. I think the OpenCV version that is installed in the ROS containers (at least the melodic) doe not come with gstreamer support, no?

mzahana, 2021-06-23:
@dusty-nv I forgot to mention that my base container is with l4t version `r32.4.3` 
I am not sure if the newer version have this issue resolved.

dusty-nv, 2021-06-23:
ROS installs different version of OpenCV I believe, which is probably not built with GStreamer.   

You could try installing the version of OpenCV from JetPack (which has GStreamer support) into the container like I do in the l4t-ml container:

* https://github.com/dusty-nv/jetson-containers/blob/6333e058495e689fcceebb6bc04eb72b5ab43893/Dockerfile.ml#L76
* https://github.com/dusty-nv/jetson-containers/blob/6333e058495e689fcceebb6bc04eb72b5ab43893/scripts/docker_build_ml.sh#L135

mzahana, 2021-06-24:
@dusty-nv Thanks! That solved the problem!