dusty-nv/jetson-containers #73: Building Ros2-desktop image

CollinNHays, 2021-07-13:
I am attempting to build Ros2-desktop by changing the the ROS_PKG argument from "ros_base" to "desktop" as done in #38 . I have attempted to do this with both Foxy and Galactic and am running into the following error.

--- stderr: sdl2_vendor
CMake Error at cmake/Modules/Findsdl2_custom.cmake:9 (target_link_libraries):
  Cannot specify link libraries for target "SDL2::SDL2" which is not built by
  this project.
Call Stack (most recent call first):
  CMakeLists.txt:12 (find_package)

---
Failed   <<< sdl2_vendor [4.70s, exited with code 1]

I can see that rosdep successfully installs libsdl2-2.0-0 and libsdl2-dev.
This is the version of sdl2_vendor used for Foxy.

    local-name: joystick_drivers/sdl2_vendor
    uri: https://github.com/ros2-gbp/joystick_drivers-release.git
    version: release/foxy/sdl2_vendor/3.0.0-2

dusty-nv, 2021-07-13:
I haven't been able to build Foxy desktop either, but Dan Pollock from the forums has:

* https://forums.developer.nvidia.com/t/ros2-foxy-zed/183451
* https://forums.developer.nvidia.com/t/script-to-install-ros2-foxy-desktop-on-agx-xavier-jetpack-4-5-1/182772

CollinNHays, 2021-07-19:
The ROS2-foxy-zed is only a ros_base image.
The script by Dan Pollock has some errors in it and doesn't work either.
I spent some time troubleshooting and was able to get foxy desktop to build in a docker container by combining your foxy Dockerfile with the script by Dan Pollock.

I also included some sensor drivers for Ouster, RoverRobotics and VectorNav in the ~/ros2_ws.

Works and tested on fresh install of Jetpack 4.5.0.
1.
Change Dokcer Default Runtime:
To enable access to the CUDA compiler (nvcc) during docker build operations, add "default-runtime": "nvidia" to your /etc/docker/daemon.json configuration file before attempting to build the containers:

{​​​​​​​
    "runtimes": {​​​​​​​
        "nvidia": {​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​
            "path": "nvidia-container-runtime",
            "runtimeArgs": []
        }​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​
    }​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​,

    "default-runtime": "nvidia"
}​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​​
You will then want to restart the Docker service or reboot your system before proceeding.

2.
Clone dusty-nv docker container scripts:
git clone https://github.com/dusty-nv/jetson-containers
cd jetson-containers
3.
Download files:
[Ros2-foxy-desktop-CollinNHays.zip](https://github.com/dusty-nv/jetson-containers/files/6841789/Ros2-foxy-desktop-CollinNHays.zip)

4.
Replace files:
In the jetson-containers directory replace "Dockerfile.ros.roxy" with the respective file downloaded above.
In the jetson-containers/scripts directory replace "docker_run.sh" withe the respective file downloaded above.
In the jetson-containers/packages directory replace "ros_entrypoint.sh" withe the respective file downloaded above.
5.
Build the container:
./scripts/docker_build_ros.sh foxy desktop
You may have to attempt to build multiple times if any of the downloads fail in the process.
This takes over an hour.
6.
Run the container:
./scripts/docker_run.sh -c ros:foxy-desktop-l4t-r32.5.1
Ctrl + d to exit and stop the container.

closed 2021-07-19 by CollinNHays