dusty-nv/jetson-containers #74: docker: Error response from daemon: OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545:

Marietto2008, 2021-07-16:
root@zi-desktop:~/Desktop/zi/Work/Android/Virt/dockers/arm64# git clone https://github.com/dusty-nv/jetson-containers

Cloning into ‘jetson-containers’…
remote: Enumerating objects: 452, done.
remote: Counting objects: 100% (33/33), done.
remote: Compressing objects: 100% (24/24), done.
remote: Total 452 (delta 15), reused 22 (delta 9), pack-reused 419
Receiving objects: 100% (452/452), 101.42 KiB | 555.00 KiB/s, done.
Resolving deltas: 100% (266/266), done.

root@zi-desktop:~/Desktop/zi/Work/Android/Virt/dockers/arm64# cd jetson-containers

root@zi-desktop:~/Desktop/zi/Work/Android/Virt/dockers/arm64/jetson-containers# scripts/docker_run.sh -c nvcr.io/nvidia/l4t-pytorch:r32.4.4-pth1.6-py3

localuser:root being added to access control list
xauth: file /tmp/.docker.xauth does not exist
Unable to find image ‘nvcr.io/nvidia/l4t-pytorch:r32.4.4-pth1.6-py3’ locally
r32.4.4-pth1.6-py3: Pulling from nvidia/l4t-pytorch
e74fe6ef6bd6: Pulling fs layer
7dcdd1c8f1d2: Pulling fs layer
148ea20d31e0: Pulling fs layer
fbc4cd4d050b: Pulling fs layer
a21b0b3d8206: Pulling fs layer
4ba0c94f9855: Pull complete
6fb19c1062d0: Pull complete
84ff17ad4b18: Pull complete
5ac903fdc4a8: Pull complete
ecf00917e120: Pull complete
30d000a9cd22: Pull complete
a26b515ffe8f: Pull complete
a199cb2dd71e: Pull complete
c4f4e0f882d3: Pull complete
3e956de9ea4b: Pull complete
e26b78d1aaed: Pull complete
ac42496d0bc2: Pull complete
7db2983f5802: Pull complete
f313550a9e60: Pull complete
a742b8e794d7: Pull complete
628353978f08: Pull complete
7adc69170422: Pull complete
e0d1fc183e52: Pull complete
948ac0abde32: Pull complete
adfb60e46ee1: Pull complete
863e90653906: Pull complete
3e9098193eb0: Pull complete
c8725fd1629c: Pull complete
97d7d811f1b0: Pull complete
Digest: sha256:05c1f230887ba3eeb6d24d7850d84d17d306f471d9c0c3a33a6e73bb2defe719
Status: Downloaded newer image for nvcr.io/nvidia/l4t-pytorch:r32.4.4-pth1.6-py3

docker: Error response from daemon: OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init caused: Running hook #0:: error running hook: exit status 1, stdout: , stderr: exec command: [/usr/bin/nvidia-container-cli --load-kmods configure --ldconfig=@/sbin/ldconfig.real --device=all --compute --compat32 --graphics --utility --video --display --pid=25195 /var/lib/docker/overlay2/d059c71d32774ffbc53e591ffd2abb0b1d2c805b6e8bb8d7af3b6365bab3186c/merged]
nvidia-container-cli: mount error: file creation failed: /var/lib/docker/overlay2/d059c71d32774ffbc53e591ffd2abb0b1d2c805b6e8bb8d7af3b6365bab3186c/merged/usr/lib/aarch64-linux-gnu/libnvidia-fatbinaryloader.so.440.18: file exists: unknown.

dusty-nv, 2021-07-16:
Hi @Marietto2008, what is your version of JetPack-L4T if you check `/etc/nv_tegra_release`?

You can get this error if you run a container on a different version of L4T than the container was built against.  You should run a tag of the container for your version of JetPack-L4T.

Marietto2008, 2021-07-16:
JetPack 4.5.1 / L4T 32.5

dusty-nv, 2021-07-16:
Try `nvcr.io/nvidia/l4t-pytorch:r32.5.0-pth1.6-py3` instead

Marietto2008, 2021-07-16:
excusme. I suszpect that I gave you the wrong version. the correct version is :

R32 (release), REVISION: 5.1, GCID: 27362550, BOARD: t210ref, EABI: aarch64, DATE: Wed May 19 18:07:59 UTC 2021

dusty-nv, 2021-07-16:
L4T R32.5.1 and R32.5.0 share the same container, because there weren’t substantial changes to the filesystem in L4T R32.5.1. So on L4T R32.5.1 you can still use nvcr.io/nvidia/l4t-pytorch:r32.5.0-pth1.6-py3

closed 2025-02-21 by johnnynunez