dusty-nv/jetson-containers #75: How to install opencv on NVIDIA L4T TensorFlow container

thasin96, 2021-07-27:
I have pulled `nvcr.io/nvidia/l4t-tensorflow:r32.5.0-tf1.15-py3` image. By default it has tensorflow, numpy, tensorrt. I want to install `opencv` as well. If i try  to install opencv ussing pip, when I import the package, it throws core dumped error. What is the right way to install `opencv` on the above container.

dusty-nv, 2021-07-27:
Hi @thasin96, are you able to `import numpy`?  If you get the same segfault, try running `export OPENBLAS_CORETYPE=ARMV8` from your terminal.  There is some bug in newer numpy, and it may be possible that pip upgraded numpy.

Alternatively, you can instead install OpenCV in container through apt:  `apt-get update && apt-get install -y python3-opencv`
I would exit and restart the container before doing this to reset the changes you made previously.

You can also install OpenCV the way I am doing in l4t-ml dockerfile:
https://github.com/dusty-nv/jetson-containers/blob/d1b4e40af1b29d19d4658de1e745ca273bbd86d9/Dockerfile.ml#L112

This way installs pre-built debian packages of OpenCV that I compiled with CUDA/cuDNN support.

thasin96, 2021-07-27:
Hi @dusty-nv , the problem was  due to pip upgradation. It installed incompatible numpy. I have resolved the issue with installing proper numpy version and now it works. 

I have gone through your hello world ai repo. It's interesting. I am currently stuck at converting uff model file to tensorrt engine. Any lead, how to convert uff model <a href="https://github.com/dusty-nv/jetson-inference/releases/tag/model-mirror-190618">files</a> to tensorrt engine file?

dusty-nv, 2021-07-27:
> I am currently stuck at converting uff model file to tensorrt engine. Any lead, how to convert uff model [files](https://github.com/dusty-nv/jetson-inference/releases/tag/model-mirror-190618) to tensorrt engine file?

This tool is how I converted the SSD-Mobilenet TensorFlow model to UFF:   https://github.com/AastaNV/TRT_object_detection

However I have found PyTorch->ONNX to be much easier and more reliable.   I believe TensorFlow has support for ONNX too.

thasin96, 2021-07-27:
Thanks for the response. I appreciate. Anyhow, I want to convert uff files to tensorrt engine files.

dusty-nv, 2021-07-27:
The trtexec tool can do it or the repo I linked to above shows it done from Python.

jetson-inference can do it, like in detectNet.cpp, but it requires that you hardcode the expected input image dimensions of the UFF model.  It is not meant to support general UFF models in jetson-inference

thasin96, 2021-07-27:
Oh okay. Thanks I will have a look.

closed 2021-07-27 by thasin96