dusty-nv/jetson-containers #78: CMake OpenCV dont found path to libraries CUDA

xentilzax, 2021-07-30:
Hi!
I have error build docker image from file Dockerfile.opencv.
CMake OpenCV dont found path to libraries CUDA
```
-- Configuring incomplete, errors occurred!
See also "/opt/opencv/build/CMakeFiles/CMakeOutput.log".
See also "/opt/opencv/build/CMakeFiles/CMakeError.log".
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_cublas_LIBRARY (ADVANCED)
    linked by target "opencv_cudev" in directory /opt/opencv_contrib/modules/cudev
    linked by target "opencv_core" in directory /opt/opencv/modules/core
    linked by target "opencv_cudaarithm" in directory /opt/opencv_contrib/modules/cudaarithm
    linked by target "opencv_cudaarithm" in directory /opt/opencv_contrib/modules/cudaarithm

```

dusty-nv, 2021-07-30:
I think you need to set your default docker-runtime to nvidia first and then reboot:
https://github.com/dusty-nv/jetson-containers#docker-default-runtime

Also here is a link to the deb's that were built by this Dockerfile:
https://nvidia.box.com/shared/static/5v89u6g5rb62fpz4lh0rz531ajo2t5ef.gz

Dockerfile.opencv is for building the debs for OpenCV (with CUDA enabled)

xentilzax, 2021-08-01:
Thank you. It helped me. Problem is solved

closed 2021-08-01 by xentilzax