dusty-nv/jetson-containers #82: fatal error: helper_cuda.h: No such file or directory, when building Dockerfile.ros.slam

mvanlobensels, 2021-08-09:
I get the following error when building the Dockerfile.ros.slam:

[image]

The error occurs when the following RUN command installs ORB_SLAM2_CUDA, which in turn executes the build.sh script:

[image]

In the build scripts from ORB_SLAM2_CUDA, I tried including the missing file without much luck. Any help would be much appreciated!

dusty-nv, 2021-08-09:
Hi @mvanlobensels, have you set your Docker default-runtime to nvidia?  https://github.com/dusty-nv/jetson-containers#docker-default-runtime

Also, can you run this command on your Jetson device (outside of container) to confirm you have `helper_cuda.h` on your system?

``` bash
$  find /usr -name 'helper_cuda.h'
/usr/local/cuda-10.2/samples/common/inc/helper_cuda.h
```

mvanlobensels, 2021-08-10:
Hi @dusty-nv,

Yes, docker default-runtime is set to nvidia. I am building this on my own x86 laptop, which does not have CUDA installed. Should the `helper_cuda.h` file not be present inside the `nvcr.io/nvidia/l4t-base:r32.5.0` base image?

mvanlobensels, 2021-08-10:
Is there a prebuilt docker image of the ros.foxy docker with SLAM available? Basically `./scripts/docker_build_ros.sh --distro foxy --with-slam`.

dusty-nv, 2021-08-10:
> I am building this on my own x86 laptop

Ah, this is untested for x86, these containers are for Jetson.

> Is there a prebuilt docker image of the ros.foxy docker with SLAM available? Basically `./scripts/docker_build_ros.sh --distro foxy --with-slam`.

Yes, you can check the tags here:  https://hub.docker.com/repository/docker/dustynv/ros/tags?page=1&ordering=last_updated&name=slam
These are for Jetson/ARM64

mvanlobensels, 2021-08-10:
Thank you! I somehow missed those. That will save me a lot of time.

closed 2021-08-10 by mvanlobensels