dusty-nv/jetson-containers #83: Adding adafruit_servokit librabry to exiisting jetson-inference container build

mmatesic01, 2021-08-14:
Hi, 

I am sucessfully able to use container with vscode so I can write code and test. 
But existing build hasn't got some libaries that I would like to use.
How can I premamnetly add "adafruit_servokit" library to my jetson-inference container build?

Thank you.

dusty-nv, 2021-08-14:
I recommend creating your own Dockerfile to install it and using jetson-inference as the base container.

Or you can install it manually, and then use 'docker commit' to save it to a new tag.

mmatesic01, 2021-08-19:
Hi again,

1.) I was able to build on your container by adding this to Docker file also based this setup source:
[](https://yunusmuhammad007.medium.com/install-driver-pcf9685-servo-controller-in-jetson-nano-41b42795ab95)

Added to Dockerfile
# install adafruit-circuitpython-servokit
RUN usermod -aG i2c root
RUN groupadd -f -r gpio
RUN usermod -a -G gpio root
RUN pip3 install adafruit-circuitpython-servokit

2.) Also I run container with added --device /dev/i2c-1 in run.sh
sudo docker run --runtime nvidia -it --rm --network host **--device /dev/i2c-1** -e DISPLAY=$DISPLAY \
    -v /tmp/.X11-unix/:/tmp/.X11-unix \
    -v /tmp/argus_socket:/tmp/argus_socket \
    -v /etc/enctune.conf:/etc/enctune.conf \
    $V4L2_DEVICES $DATA_VOLUME $USER_VOLUME \
    $CONTAINER_IMAGE $USER_COMMAND

3.) When I run this command  "i2cdetect -y -r 1" out or within the container it shows me that PCF9685 controler is visible inside container.
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- --

4.) I run this simple code (servos move when I run outside container):

_from adafruit_servokit import ServoKit

_kit = ServoKit(channels=16)
pan = 90
tilt = 90

kit.servo[0].angle = pan
kit.servo[1].angle = tilt_

5.) But when I running the code inside container, I get the next message:
__Traceback (most recent call last):
File "/home/marko/Desktop/pyProg/NVIDIA/servotest.py", line 1, in
from adafruit_servokit import ServoKit
File "/usr/local/lib/python3.6/dist-packages/adafruit_servokit.py", line 35, in
import board
File "/usr/local/lib/python3.6/dist-packages/board.py", line 33, in
from adafruit_blinka.agnostic import board_id, detector
File "/usr/local/lib/python3.6/dist-packages/adafruit_blinka/agnostic/init.py", line 18, in
chip_id = detector.chip.id
File "/usr/local/lib/python3.6/dist-packages/adafruit_platformdetect/chip.py", line 350, in getattr
if self.id == attr:
File "/usr/local/lib/python3.6/dist-packages/adafruit_platformdetect/chip.py", line 350, in getattr
if self.id == attr:
File "/usr/local/lib/python3.6/dist-packages/adafruit_platformdetect/chip.py", line 350, in getattr
if self.id == attr:
[Previous line repeated 320 more times]
File "/usr/local/lib/python3.6/dist-packages/adafruit_platformdetect/chip.py", line 154, in id
self._chip_id = self._linux_id()
File "/usr/local/lib/python3.6/dist-packages/adafruit_platformdetect/chip.py", line 235, in linux_id
hardware = self.detector.get_cpuinfo_field("Hardware")
File "/usr/local/lib/python3.6/dist-packages/adafruit_platformdetect/init.py", line 49, in get_cpuinfo_field
with open("/proc/cpuinfo", "r") as infile:
File "/usr/lib/python3.6/codecs.py", line 309, in init
IncrementalDecoder.init(self, errors)
RecursionError: maximum recursion depth exceeded_

Running this code outside container works.
What else am I missing in my container?

closed 2025-02-21 by johnnynunez