dusty-nv/jetson-containers #84: unauthorized issue when building ros2 foxy with script

marklee56826, 2021-08-16:
Hi Dusty,
I have the same issue as jvschw commented on Oct 26, 2020
......
Step 1/22 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.4.3
Step 2/22 : FROM ${BASE_IMAGE}
unauthorized: authentication required

According to your instructions, I git cloned the newest repo: https://github.com/dusty-nv/jetson-containers and then cd jetson-container; used the right command ./scripts/docker_build_ros.sh --distro foxy. However, I still got the error. I need your help.
configuration: l4t-r32.61, ubuntu 18.04, Jetson Xavier NX
Thank you.

dusty-nv, 2021-08-16:
Hi @marklee56826, can you pull the latest from the repo and try again?

I just pushed a change in https://github.com/dusty-nv/jetson-containers/commit/d58ce7eb0afbb3c2706fc62d26f69e7055384484 which I hope fixes it.

marklee56826, 2021-08-17:
Thank you for your kind reply.

closed 2021-08-17 by marklee56826