dusty-nv/jetson-containers #86: OCI runtime create failed: container_linux.go:380: starting container process caused: process_linu                 x.go:545: container init caused: Running hook #0:: error running hook: signal: segmentation fault (core dumped), stdout: , stderr:: un                 known.

chaibi-ali, 2021-08-18:
when i do write this in terminal: sudo docker run --rm --gpus all nvidia/cuda:11.0-base nvidia-smi 
i have this error : ** OCI runtime create failed: container_linux.go:380: starting container process caused: process_linu                 x.go:545: container init caused: Running hook #0:: error running hook: signal: segmentation fault (core dumped), stdout: , stderr:: un                 known.**

dusty-nv, 2021-08-19:
This is an x86 container, not an aarch64 container for JetPack-L4T.  Please run one of the l4t containers like l4t-base, l4t-pytorch, l4t-ml, ect.  l4t-base has CUDA/cuDNN/TensorRT

chaibi-ali, 2021-08-20:
am using vm ubuntu 20.04 LTS i dont now how to do that ...

dusty-nv, 2021-08-20:
> am using vm ubuntu 20.04 LTS i dont now how to do that ...

Sorry, if you are trying to run an x86 container on an x86 system, this question is unrelated to this repo.  This repo is for building containers for Jetson/aarch64.

closed 2021-09-03 by chaibi-ali

menghe999, 2021-10-17:
I also encountered a similar problem
exec this cmd  `service docker status` , result as follow

`
● docker.service - Docker Application Container Engine
   Loaded: loaded (/usr/lib/systemd/system/docker.service; disabled; vendor preset: disabled)
   Active: active (running) since Sun 2021-10-17 11:11:12 CST; 8min ago
     Docs: https://docs.docker.com
 Main PID: 455648 (dockerd)
    Tasks: 21
   Memory: 31.0M
   CGroup: /system.slice/docker.service
           ├─455648 /usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock
           └─455794 /usr/bin/docker-proxy -proto tcp -host-ip 0.0.0.0 -host-port 5601 -container-ip 172.18.0.2 -container-port 5601

Oct 17 11:11:11 iZbp1bsgdsxr870r0lp9wtZ dockerd[455648]: time="2021-10-17T11:11:11.294978357+08:00" level=info msg="Default bridge (docker0) is assigned with an IP address 172.18.0.0/16. Daemon option --bip can be used to set a preferred IP address"
Oct 17 11:11:11 iZbp1bsgdsxr870r0lp9wtZ dockerd[455648]: time="2021-10-17T11:11:11.639616085+08:00" level=error msg="stream copy error: reading from a closed fifo"
Oct 17 11:11:11 iZbp1bsgdsxr870r0lp9wtZ dockerd[455648]: time="2021-10-17T11:11:11.643133778+08:00" level=error msg="stream copy error: reading from a closed fifo"
Oct 17 11:11:11 iZbp1bsgdsxr870r0lp9wtZ dockerd[455648]: time="2021-10-17T11:11:11.802749887+08:00" level=error msg="c5b9b46578c77a51d838c93baeb294db32c5017d9230a142db26b1aeccb9605a cleanup: failed to delete container from containerd: no such container"
Oct 17 11:11:11 iZbp1bsgdsxr870r0lp9wtZ dockerd[455648]: time="2021-10-17T11:11:11.802782345+08:00" level=error msg="failed to start container" container=c5b9b46578c77a51d838c93baeb294db32c5017d9230a142db26b1aeccb9605a error="OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init c
Oct 17 11:11:11 iZbp1bsgdsxr870r0lp9wtZ dockerd[455648]: time="2021-10-17T11:11:11.873286216+08:00" level=info msg="Loading containers: done."
Oct 17 11:11:11 iZbp1bsgdsxr870r0lp9wtZ dockerd[455648]: time="2021-10-17T11:11:11.912254127+08:00" level=info msg="Docker daemon" commit=79ea9d3 graphdriver(s)=overlay2 version=20.10.9
Oct 17 11:11:11 iZbp1bsgdsxr870r0lp9wtZ dockerd[455648]: time="2021-10-17T11:11:11.912331759+08:00" level=info msg="Daemon has completed initialization"
Oct 17 11:11:12 iZbp1bsgdsxr870r0lp9wtZ systemd[1]: Started Docker Application Container Engine.
Oct 17 11:11:12 iZbp1bsgdsxr870r0lp9wtZ dockerd[455648]: time="2021-10-17T11:11:12.065941762+08:00" level=info msg="API listen on /var/run/docker.sock"
`
now , docker couldn't use, tips this
`
[root@iZbp1bsgdsxr870r0lp9wtZ ~]# docker ps
Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?

`
Has anyone encountered such a problem