dusty-nv/jetson-containers #87: could not build python packages in docker containers

marklee56826, 2021-08-19:
Hi Dusty,
I pulled two of your images: ROS2 Foxy both R32.6.1 and R32.5.0. When trying to build a package with “colcon build”, I always failed:
…Dist.py:700 UserWarning: Usage of dash-separated ‘script-dir’ will not be supported in future versions…
……
1 package had stderr output: my_package.
My system: l4t-r32.61, ubuntu 18.04, Jetson Xavier NX, python setuptools version:'57.4.0'.  
What I should do to remove the stderr?
Thank you.

dusty-nv, 2021-08-19:
I don't believe this actually failed.  Was there other errors?   If not, I would ignore it.

marklee56826, 2021-08-19:
Thank you for your kind reply. 
I just described the ros2 installations process on my NX board with the newest Jetpack 32.6.1 but I did not mean there are really issues in the docker image or in scripts. Therefore, I will move my questions into Nvidia forums and close the issue, thank you.

closed 2021-08-19 by marklee56826