dusty-nv/jetson-containers #88: Tensorflow wheel for jetpack

alborotogarcia, 2021-08-23:
Please could you provide on https://github.com/dusty-nv/jetson-containers/blob/d58ce7eb0afbb3c2706fc62d26f69e7055384484/scripts/docker_build_ml.sh#L135 a referefence for the bazel file or a script that builds that wheel?

I've successfully build 2.4.1 gpu modifying your l4t base image for ubuntu focal and cuda 10.2 so I can get rid of py3.6 and use py3.8 and pandas 1.2+

So I can get xgboost and pyarrow versions up to date.

Thanks

dusty-nv, 2021-08-23:
@alborotogarcia I don't build these wheels or have the scripts for them.  They come from:  https://docs.nvidia.com/deeplearning/frameworks/install-tf-jetson-platform/index.html#prereqs

My understanding however is that it is built relatively normally according to how you would normally build TensorFlow.

alborotogarcia, 2021-08-23:
I see, but last time I tried I was unable to build it successfully.. from the docs you can clearly see that they aimed to [drop cuda 10.2 support](https://www.tensorflow.org/install/gpu)

Same applies to rapids from 0.19 and so on..

dusty-nv, 2021-08-23:
Here is an older forum thread about building TensorFlow for Jetson:  https://devtalk.nvidia.com/default/topic/1055131/jetson-agx-xavier/building-tensorflow-1-13-on-jetson-xavier/

Unfortunately I am unable to support because I've never built TensorFlow before.  Same for Rapids, although my understanding is that Rapids is not supported on Jetson.

alborotogarcia, 2021-08-23:
Yes, although I was able to build cudf on xavier see https://github.com/rapidsai/cuml/issues/3929#issue-909051771 
The only missing piece was cuml, here's the Rapidsai builds so far https://hub.docker.com/r/gpuci/rapidsai

@dusty-nv Could you know who is in charge of maintaining the wheels from https://developer.download.nvidia.com/compute/redist/jp/ ? 

Anyways, thanks for the help