dusty-nv/jetson-containers #97: How to open /dev/ttyTHS1 serial port on Jetson nano

kyuhyong, 2021-10-11:
I was able to run ros2 from docker image however I could not find /dev/ttyTHS1 port while running in the docker.
I see /dev/ttyTHS1 exist out of docker.
Is there a way to use this port inside the docker?
Thanks,

alma11, 2022-08-24:
Dusty, I am having the same issue---I have RxTx lines tied between Arduino Due and Jetson NX.  I successfully send data back and forth over serial (using the Jetsons ttyTHS0 line) when NOT in the docker. I have successfully run the docker to perform object detection. That said, when I try to run the Serial program from the docker, and I load the docker using the following command (using the hint you provided above):

_docker /run.sh --volume ~/OD/Model_001:/Model_001 --device=/dev/ttyTHS0_

I get the following error: 

_[Errno 2] could not open port /dev/ttyTHS0: [Errno 2] No such file or directory:  '/dev/ttyTHS0'_

Is there something obvious I am doing wrong? As a sanity check, I looked if ttyTHS0 was actually in /dev, and it is. Any help on this would really be appreciated. I am assuming this is something to do with the path to /dev/ttyTHS0.

dusty-nv, 2022-08-24:
> Is there something obvious I am doing wrong? As a sanity check, I looked if ttyTHS0 was actually in /dev, and it is.

Is it actually in `/dev` *inside* the container, or only outside?

If you are using my `scripts/docker_run.sh` script, it doesn't support a `--device` flag on the command-line - you will need to edit it and manually add it to the `sudo docker run` command near the bottom of the script.

alma11, 2022-08-25:
Dusty, to start---thank you for the helpful hint and the fast reply. I was able to resolve this per your advice. Just to be complete for someone else who might come along and have the same question---these were my steps:

1. Navigate to the docker directory located in the jetson-inference folder (might be different directory for someone else).

> _home/jetson-inference/docker/_

2. In the docker folder, you will see a _run.sh_ file. Right click it, copy it, and paste it into docker, renaming it to _run_edit.sh_. The reason I did this was I wanted to preserve the original run.sh file in case the run_edit.sh file was messed up.

3. Right click the _run_edit.sh_ file from file manager, and open with Text Editor (You might have to first select Open With Other Application to select Text Editor from a list of applications).

4. With the run_edit.sh file open, scroll down to the bottom and find _#check for V4L2 devices_.

5. Copy the entire _#check for V4L2 devices_ section and paste it just before the _#run the container_ section. You will now edits to this section. Change the comment to _#check for ttyTHS0 devices_ (for easy recall). Then go through and change all the V4L2_DEVICES text to TTYTHS_DEVICES, as well as other things shown below in italic (BTW, the below code segment is ready to be copied and pasted into the previously made run_edit.sh file). Compare the below text with what already exits to see the differences.

> # check for ttyTHS0 devices
> _TTYTHS_DEVICES_=" "
> 
> for i in {0..9}
> do
> 	if [ -a "_/dev/ttyTHS_$i" ]; then
> 		_TTYTHS_DEVICES_="$_TTYTHS_DEVICES_ _--device /dev/ttyTHS_$i "
> 	fi
> done
> 
> echo "_TTYTHS_DEVICES_:  $_TTYTHS_DEVICES_"

6. Finally, in the _#run the container section_, add in the $TTYTHS_DEVICES. Compare the below text with what already exits to see the differences.

> if [ $ARCH = "aarch64" ]; then
> 
> 	sudo docker run --runtime nvidia -it --rm \
> 		--network host \
> 		-e DISPLAY=$DISPLAY \
> 		-v /tmp/.X11-unix/:/tmp/.X11-unix \
> 		-v /tmp/argus_socket:/tmp/argus_socket \
> 		-v /etc/enctune.conf:/etc/enctune.conf \
> 		$V4L2_DEVICES $TTYTHS_DEVICES $DATA_VOLUME $USER_VOLUME \
> 		$CONTAINER_IMAGE $USER_COMMAND

7. Now that a new run_edit.sh file has been created, we need to run this with docker. Note that you will also have to install Serial every time as the docker forgets this install with each new container creation. Note that the directory shown below is something I created (OD/CustomMode_001) that I wanted to start the volume from. Someone else's will differ.  I also put my python script in the CustomModel_001 folder, that is why I call it directly in the last command.

> - cd jetson-inference
> - sudo docker /run_edit.sh --volume ~/OD/CustomModel_001:/CustomModel_001
> - python3 my-detection-Serial.py

At this point you are ready to run your program---and your Serial ttyTHS0 should now work.

alma11, 2022-08-25:
Dusty-nv, I do have one more question which relates to my step '7' above. Every time I run the docker, it seems to forget I have previously installed pyserial. Meaning I have to run:

> pip3 install pyserial

Every time exit and restart docker. Do you know how to make it such that installed libraries are saved within docker upon exit and entry?

dusty-nv, 2022-08-25:
Hi @alma11, glad you were able to get your serial devices working - to save persistent changes to the container, you can use the [`docker commit`](https://docs.docker.com/engine/reference/commandline/commit/) command

closed 2025-02-21 by johnnynunez