dusty-nv/jetson-containers #104: libcuda.so: file not recognized: File truncated

malekijoo, 2021-10-31:
I am using  the image `dustynv/ros:melodic-ros-base-l4t-r32.6.1`. I set the `/etc/docker/daemon.json` as it describes in the [Read.me](https://github.com/dusty-nv/jetson-containers#docker-default-runtime), I also installed `nvidia-docker2` as its [instruction](https://github.com/dusty-nv/jetson-containers#docker-default-runtime).
My Nvidia driver is 470.63.01 and Cuda 11.4 on the host. The image has Cuda 10.2. 
When I run and compile an application called [darknet-ros](https://github.com/leggedrobotics/darknet_ros) with C++  under catkin this Error comes up. 

```
Errors     << darknet_ros:make /root/yolo_ws/logs/darknet_ros/build.make.001.log                                                                                       
/usr/lib/gcc/aarch64-linux-gnu/7/../../../aarch64-linux-gnu/libcuda.so: file not recognized: File truncated
collect2: error: ld returned 1 exit status
make[2]: *** [/root/yolo_ws/devel/.private/darknet_ros/lib/libdarknet_ros_lib.so] Error 1
make[1]: *** [CMakeFiles/darknet_ros_lib.dir/all] Error 2
make: *** [all] Error 2
```
The `LD_LIBRARY_PATH` and the `aarch64-linux-gnu` is as follow,
```
root@lab:/# ls -lah /usr/lib/x86_64-linux-gnu/libcuda*
lrwxrwxrwx 1 root root  12 Oct 26 08:21 /usr/lib/x86_64-linux-gnu/libcuda.so -> libcuda.so.1
lrwxrwxrwx 1 root root  20 Oct 26 08:21 /usr/lib/x86_64-linux-gnu/libcuda.so.1 -> libcuda.so.470.63.01
-rw-r--r-- 1 root root 23M Aug  3 20:51 /usr/lib/x86_64-linux-gnu/libcuda.so.470.63.01

root@lab:/# echo $LD_LIBRARY_PATH
/opt/ros/melodic/lib:/usr/local/cuda-10.2/targets/aarch64-linux/lib

```

In [856](https://github.com/NVIDIA/nvidia-docker/issues/856) a similar error came up which building without [GPU support] and `"default-runtime": "nvidia"` is suggested. The application I am using needs to access the GPU on the host to run and compile with C++ which removing those symlink `libcuda.so` won't help me.

I was wondering if it is a bug originating from building `dustynv/ros:melodic-ros-base-l4t-r32.6.1` or it is on my side? 
Can anyone tell me what is wrong?

camelator, 2022-05-23:
same issue here following this step by step 
https://yunusmuhammad007.medium.com/build-and-install-opencl-on-jetson-nano-10bf4a7f0e65
as result, I am not able to find a way to compile opencv within this container

wrickout, 2023-02-07:
same issue on r35.1  on noetic

dusty-nv, 2023-02-07:
> My Nvidia driver is 470.63.01 and Cuda 11.4 on the host. The image has Cuda 10.2.

These container images are built for Jetson/ARM platform and not x86.   On x86, you could either use OSRF's official ROS containers as your base container, or change the base container that I build with to something other than l4t-base.   l4t-base is for Jetson/ARM has stubs for those cuda libraries in it which is probably what is causing you that error.

closed 2025-02-21 by johnnynunez