dusty-nv/jetson-containers #105: ROS2 Galactic Container with Python3.8

fas2037, 2021-11-01:
When installing ROS2 Galactic natively it uses Python3.8. The container uses Python3.6 and therefore libraries depending on Python3.8, which could be uses in a native ROS2 Galactic environment, can't be used inside of the docker.

dusty-nv, 2021-11-01:
@fas2037 in order to get GPU acceleration, the ROS2 containers from this repo are based on l4t-base container, which uses an Ubuntu 18.04 filesystem (and 18.04 comes with Python 3.6).   It using l4t-base and 18.04 are the reason that ROS2 Foxy/Galactic are built from source as opposed to being installed from the pre-built debian packages from the ROS apt repo.  

If you don't require GPU acceleration, then you can just use the normal ROS galactic container from OSRF, which is based on 20.04 / Python 3.8 and their pre-built debian packages.

fas2037, 2021-11-02:
Okay, but as the GPU acceleration depends on Python3.6 it is not possible to run programs with GPU acceleration and Python3.8 libs, right?

dusty-nv, 2021-11-02:
> GPU acceleration depends on Python3.6 it is not possible to run programs with GPU acceleration and Python3.8 libs

GPU acceleration itself doesn't depend on Python 3.6 - GPU acceleration _from within containers_ depends on using a container derived from l4t-base.   And l4t-base is based on Ubuntu 18.04, which uses Python 3.6 by default.   So by default, it is using Python 3.6.

That isn't to say that you couldn't build the container to use Python 3.8 instead.  You can try changing my ROS Galactic dockerfile to install/use Python 3.8 instead:

https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ros.galactic

You would want to do this before ROS2 is built in the dockerfile, because you want ROS2 to get built against the version of Python that you installed.

fas2037, 2022-02-09:
So what I do is building the usual ros:galactic with python3.8 for arm64 using buildx

closed 2025-02-21 by johnnynunez