dusty-nv/jetson-containers #107: ros noetic full desktop image?

maghauke, 2021-11-03:
Any tips on how to create a docker image for jetson xavier agx, that contains ros noetic desktop full?

dusty-nv, 2021-11-03:
You can try running `scripts/docker_build_ros.sh --distro noetic --package desktop`, but sometimes there are errors building desktop for Noetic/Foxy/Galactic because these are based on 20.04 and there are library/API conflicts with the 18.04 versions.

rolfkwakkelaar, 2022-04-23:
According to http://wiki.ros.org/melodic/Installation/Ubuntu there are 'desktop' and 'desktop-full'  . Am I correct that docker_build_ros.sh does not support desktop-full ?

dusty-nv, 2022-04-25:
Hi @rolfkwakkelaar, since Melodic is based on 18.04, with JetPack 4.x you may be able to make it work by adding `desktop-full` to this line:   https://github.com/dusty-nv/jetson-containers/blob/5a53691558900f2264ba29b5996ce531d4fa8abb/scripts/docker_build_ros.sh#L58

closed 2025-02-21 by johnnynunez