dusty-nv/jetson-containers #111: cannot build noetic --with-pytorch :

arpaterson, 2021-12-02:
When I try to build using 
`./scripts/docker_build_ros --with-pytorch`

The build stops at step 2/18
`manifest for dustynv/jetson-inference:r32.5.1 not found`

I'm not sure if I am missing a step, but I haven't seen anything to suggest I am.

The _ml all built fine, and I am only interested in ros noetic.
The nano I'm using has l4t 32.5.1.

dusty-nv, 2021-12-02:
Hi @arpaterson, can you try changing this line of code to the below?

https://github.com/dusty-nv/jetson-containers/blob/56bdf7e31037b308b6ef3039175c80a64e5e9d81/scripts/docker_build_ros.sh#L140

```
BASE_IMAGE_PYTORCH="jetson-inference:r32.5.0"
```

This is to accommodate the special case where L4T R32.5.1 actually uses the R32.5.0 containers.

dusty-nv, 2021-12-03:
The SLAM dockerfile is setup for the ROS2 versions of the SLAM packages, so I haven't tried to build it on Noetic

closed 2025-02-21 by johnnynunez