dusty-nv/jetson-containers #112: Can't build Docker for ROS Foxy with SLAM

symonssoftware, 2021-12-02:
Trying to build the ROS2 Foxy Docker with SLAM so that I can run the ZED ROS wrapper that's built into that image.

Running JetPack 4.6 on a Jetson Nano 4GB.

Command executed: ./scripts/docker_build_ros.sh --distro foxy --with-slam

Error message when build fails:

CMake Error at /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:603 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:825 (_pkg_check_modules_internal)
  components/pango_windowing/CMakeLists.txt:38 (pkg_check_modules)
  CMakeLists.txt:93 (include)

-- Configuring incomplete, errors occurred!
See also "/tmp/pangolin/build/CMakeFiles/CMakeOutput.log".
See also "/tmp/pangolin/build/CMakeFiles/CMakeError.log".
The command '/bin/bash -c git clone https://github.com/stevenlovegrove/Pangolin /tmp/pangolin &&     cd /tmp/pangolin &&     mkdir build &&     cd build &&     cmake ../ &&     make -j$(nproc) &&     make install' returned a non-zero code: 1

----------------------------------------

Also, should I change the ZED_SDK_URL environment variables to match the latest ZED wrapper version, 3.6 with JetPack 4.6? (i.e. ZED_SDK_URL="https://download.stereolabs.com/zedsdk/3.6/jp46/jetsons")? Should the ZED_SDK_RUN variable be changed to something else as well?

symonssoftware, 2021-12-08:
Tried burning a clean microSD card image with JetPack 4.6 for my Nano and building the ROS2 Foxy Docker image with SLAM ...  still didn't work. Is anyone else having issues building these images?

dusty-nv, 2021-12-08:
@symonssoftware I wonder if `pkg-config` needs installed with apt in the Dockerfile.slam before pangolin gets installed (although this was not a problem before).   Note that the SLAM container I don't regularly maintain due to it's complexity and number of dependencies.

> Also, should I change the ZED_SDK_URL environment variables to match the latest ZED wrapper version, 3.6 with JetPack 4.6? (i.e. ZED_SDK_URL="https://download.stereolabs.com/zedsdk/3.6/jp46/jetsons")

Yes, I would change ZED SDK version to JetPack 4.6 version since you are building for JetPack 4.6

symonssoftware, 2021-12-08:
@dusty-nv Thanks for the response. I really don't need the SLAM capabilities in the Dockerfile. My most immediate need is just to be able to run the ZED ROS2 Foxy Wrapper on the Jetson Nano and it looks like the wrapper install was part of the SLAM Dockerfile. 

I'm trying to put together a STEM platform for my son's high-school FIRST Robotics team so that they can learn more about the use of ROS as well as some aspects of AI vision. The robot has a Raspberry Pi 4 running ROS2 Foxy connected via a MicroROS agent to a Teensy 4.1 board for motor and sensor control. I also have a Jetson Nano 4GB board and the Stereolab's ZED 2i camera  mounted on the bot.

If there's a more straightforward way to get the wrapper up and running, I'd really appreciate the help.

dusty-nv, 2021-12-08:
OK, so if you just want the ZED stuff, I would remove basically all the other packages from that Dockerfile than the ZED stuff.

Basically remove lines 45 through 235 from Dockerfile.slam

symonssoftware, 2021-12-08:
@dusty-nv Thanks, I'll give that a try.

symonssoftware, 2021-12-09:
@dusty-nv I modified the SLAM dockerfile as you suggested. It looks like it's failing when trying to setup the ZED ROS2 wrapper. See the build output below:

--------------

Cloning into 'src/slam/zed-ros2-wrapper'...
Get:1 http://packages.ros.org/ros2/ubuntu bionic InRelease [4,680 B]
Get:2 https://apt.kitware.com/ubuntu bionic InRelease [11.0 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports bionic InRelease [242 kB]
Get:4 http://packages.ros.org/ros2/ubuntu bionic/main arm64 Packages [658 kB]
Get:5 https://apt.kitware.com/ubuntu bionic/main arm64 Packages [22.1 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease [88.7 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease [74.6 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease [88.7 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports bionic/restricted arm64 Packages [572 B]
Get:10 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 Packages [1,285 kB]
Get:11 http://ports.ubuntu.com/ubuntu-ports bionic/universe arm64 Packages [11.0 MB]
Get:12 http://ports.ubuntu.com/ubuntu-ports bionic/multiverse arm64 Packages [153 kB]
Get:13 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 Packages [1,748 kB]
Get:14 http://ports.ubuntu.com/ubuntu-ports bionic-updates/multiverse arm64 Packages [5,317 B]
Get:15 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe arm64 Packages [1,989 kB]
Get:16 http://ports.ubuntu.com/ubuntu-ports bionic-updates/restricted arm64 Packages [4,427 B]
Get:17 http://ports.ubuntu.com/ubuntu-ports bionic-backports/universe arm64 Packages [11.0 kB]
Get:18 http://ports.ubuntu.com/ubuntu-ports bionic-backports/main arm64 Packages [11.2 kB]
Get:19 http://ports.ubuntu.com/ubuntu-ports bionic-security/restricted arm64 Packages [3,750 B]
Get:20 http://ports.ubuntu.com/ubuntu-ports bionic-security/multiverse arm64 Packages [2,824 B]
Get:21 http://ports.ubuntu.com/ubuntu-ports bionic-security/main arm64 Packages [1,360 kB]
Get:22 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe arm64 Packages [1,291 kB]
Fetched 20.1 MB in 5s (4,055 kB/s)
Reading package lists...
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
zed_wrapper: No definition of [xacro] for OS version [bionic]
zed_components: No definition of [diagnostic_updater] for OS version [bionic]
zed_ros2: Cannot locate rosdep definition for [zed_interfaces]
The command '/bin/bash -c source ${ROS_ENVIRONMENT} &&     cd ${ROS_ROOT} &&     mkdir -p src/slam &&     git clone https://github.com/stereolabs/zed-ros2-wrapper src/slam/zed-ros2-wrapper &&     apt-get update &&     rosdep install -y       --ignore-src        --from-paths src/slam 	  --rosdistro ${ROS_DISTRO} 	  --skip-keys "rtabmap find_object_2d Pangolin libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" &&     rm -rf /var/lib/apt/lists/* &&     apt-get clean &&     colcon build        --merge-install 	  --base-paths src/slam/zed-ros2-wrapper 	  --cmake-args=-DCMAKE_BUILD_TYPE=Release &&     rm -rf ${ROS_ROOT}/src &&     rm -rf ${ROS_ROOT}/logs &&     rm -rf ${ROS_ROOT}/build' returned a non-zero code: 1

dusty-nv, 2021-12-09:
Hi @symonssoftware, sorry I forgot that I have the pre-built foxy-slam container image for JetPack 4.6 already on DockerHub:

``` bash
sudo docker pull dustynv/ros:foxy-slam-l4t-r32.6.1
```

This includes the ZED SDK.  Are you able to pull/run that?

symonssoftware, 2021-12-09:
@dusty-nv Thanks for the help. That worked!!!

Just in case anyone else is looking at this issue and wants to do the same thing, I also did have to provide access to the USB ports that the ZED 2i camera uses when the Docker image is launched like this:

$sudo docker run --runtime nvidia -it --rm --network host --privileged -v /dev:/dev docker.io/dustynv/ros:foxy-slam-l4t-r32.6.1

dusty-nv, 2021-12-09:
OK great, glad to hear it!   Sorry I forgot about that earlier... :)

symonssoftware, 2021-12-09:
@dusty-nv No problem. I'm new to a lot of this and I appreciate you walking me through the process of getting this up and running.

javadan, 2022-01-30:
I'm at a similar spot now...

```
./scripts/docker_build_ros.sh --distro foxy --with-slam

(building)
...
(building)

Successfully built 21ec8e41dd0d
Successfully tagged ros:foxy-pytorch-l4t-r32.5.0

Building container ros:foxy-slam-l4t-r32.5.0
BASE_IMAGE=ros:foxy-pytorch-l4t-r32.5.0

Building ros:foxy-slam-l4t-r32.5.0 container...

```

It just stays there, for an hour until I CTRL+C.

I can pull the one from dockerhub though, so will continue with that for now.

dusty-nv, 2022-03-17:
I don't really maintain the SLAM docker anymore, perhaps I should just mark it for removal.  It appears that in your build, it is missing Eigen or can't find it for some reason.

medphisiker, 2022-03-17:
> I don't really maintain the SLAM docker anymore, perhaps I should just mark it for removal. It appears that in your build, it is missing Eigen or can't find it for some reason.

Thank you )

closed 2025-02-21 by johnnynunez