dusty-nv/jetson-containers #113: No audio backend is available for torchaudio

znmeb, 2021-12-08:
It looks like there's no audio backend for `torchaudio` installed on the PyTorch image:

```
❯ docker run -it nvcr.io/nvidia/l4t-pytorch:r32.6.1-pth1.9-py3 /bin/bash
root@e30aa8479fa3:/# python3
Python 3.6.9 (default, Jan 26 2021, 15:33:00)
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> import torchaudio
/usr/local/lib/python3.6/dist-packages/torchaudio-0.9.0a0+33b2469-py3.6-linux-aarch64.egg/torchaudio/backend/utils.py:67: UserWarning: No audio backend is available.
  warnings.warn('No audio backend is available.')
>>> torchaudio.set_audio_backend("sox_io")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/torchaudio-0.9.0a0+33b2469-py3.6-linux-aarch64.egg/torchaudio/backend/utils.py", line 44, in set_audio_backend
    f'Backend "{backend}" is not one of '
RuntimeError: Backend "sox_io" is not one of available backends: [].
>>>
```

dusty-nv, 2021-12-08:
Hmm it appears that sox package is installed... https://github.com/dusty-nv/jetson-containers/blob/56bdf7e31037b308b6ef3039175c80a64e5e9d81/Dockerfile.pytorch#L92

so I wonder why torchaudio is not picking it up...

znmeb, 2021-12-08:
```
$ apt-cache search libsox
libsoxr-dev - High quality 1D sample-rate conversion library (development files)
libsoxr-lsr0 - High quality 1D sample-rate conversion library (libsamplerate bindings)
libsoxr0 - High quality 1D sample-rate conversion library
libsox-dev - Development files for the SoX library
libsox-fmt-all - All SoX format libraries
libsox-fmt-alsa - SoX alsa format I/O library
libsox-fmt-ao - SoX Libao format I/O library
libsox-fmt-base - Minimal set of SoX format libraries
libsox-fmt-mp3 - SoX MP2 and MP3 format library
libsox-fmt-oss - SoX OSS format I/O library
libsox-fmt-pulse - SoX PulseAudio format I/O library
libsox3 - SoX library of audio effects and processing
sox - Swiss army knife of sound processing
```

You may need `libsox-dev`and `libsox-fmt-all` and possibly `libsoxr-dev` as well. `sox` is a main program; the "-dev" files provide headers if `torchaudio` needs them to compile using the libraries, and "-fmt-all" brings in libraries for things like FLAC. `libsoxr-dev` is for sample rate conversion,

dusty-nv, 2022-02-18:
Confirmed this is fixed in commit https://github.com/dusty-nv/jetson-containers/commit/96fb7f2fa31fd6494d5ce6093377f2ccf4af1dbc

The issue ended up being that `BUILD_SOX=1` environment variable needed set because torchaudio builds these from source, as opposed to using the debian packages (so installing those was having no effect)

closed 2022-02-18 by dusty-nv