dusty-nv/jetson-containers #114: How to install package dependencies using existing install?

vectorcrumb, 2021-12-16:
Hi, I'm trying to install additional packages on top of the ROS2 foxy image with the attached Dockerfile at the end, however I get the following error when running `rosdep install`:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
image_transport_plugins: No definition of [theora_image_transport] for OS version [bionic]
urg_node: No definition of [urg_c] for OS version [bionic]
```
I'm trying to run `rosinstall_generator` with the `--deps` option to reduce download and build times, however this means that the dependencies for my packages aren't pulled into the `.rosinstall` file. When trying to run `rosdep install`, the additional dependencies aren't found because the Ubuntu image is bionic and not focal. Is there any way to consider the packages installed in /opt/ros/foxy from the original Dockerfile to reduce the amount of downloaded packages when using `rosinstall_generator --deps ...`?

```
ARG BASE_IMAGE=dustynv/ros:foxy-ros-base-l4t-r32.6.1
FROM ${BASE_IMAGE}

ENV ROS_DOMAIN_ID=0
ENV ROS_WS=/root/dev_ws

RUN apt-get update && \
    apt-get install -y --no-install-recommends \
		nano \
		software-properties-common \
    && rm -rf /var/lib/apt/lists/* \
    && apt-get clean

# Install librealsense2 SDK
RUN apt-key adv --keyserver keyserver.ubuntu.com --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
RUN add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u && \
	apt-get update && \
	apt-get install librealsense2-utils -y && \
	apt-get install librealsense2-dev -y && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean
# Install extra packages
RUN mkdir -p ${ROS_WS}/src && \
    cd ${ROS_WS} && \
    source ${ROS_ROOT}/install/setup.bash && \
    rosinstall_generator --rosdistro ${ROS_DISTRO} ${ROS_PKG} \
		ament_cmake \
                diagnostic_updater \
                image_transport_plugins \ 
                urg_node_msgs \
		xacro \
		urg_node \
		usb_cam \
        > ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall && \
    cat ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall && \
    vcs import src < ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall && \
    apt-get update && \
    cd ${ROS_WS} && \
    rosdep update && \
    rosdep install -y \
    	  --ignore-src \
          --from-paths src \	
	  --rosdistro ${ROS_DISTRO} \
	  --skip-keys "libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean && \
    # build it!
    colcon build --merge-install && \
    # remove build files
    rm -rf ${ROS_WS}/src && \
    rm -rf ${ROS_WS}/logs && \
    rm -rf ${ROS_WS}/build && \
    rm ${ROS_WS}/*.rosinstall

COPY ros_entrypoint.sh /ros_entrypoint.sh
RUN echo 'source ${ROS_WS}/install/setup.bash' >> /root/.bashrc

ENTRYPOINT ["/ros_entrypoint.sh"]
CMD ["bash"]
WORKDIR /root/

```

dusty-nv, 2021-12-16:
Hi @vectorcrumb, here is how I build additional packages/dependencies on top of the ROS2 containers:

https://github.com/dusty-nv/jetbot_ros/blob/e2bd28b3e182d772ef051f1cc0e5c42c9fb4f1ec/Dockerfile#L72

Note the use of `export ROS_PACKAGE_PATH=${AMENT_PREFIX_PATH}` and `--exclude RPP`

vectorcrumb, 2021-12-16:
Thanks dusty, that worked great! One more question: how do you select the `--skip-keys` arguments you use? Are you asking rosdep to avoid installing packages available throught `apt` that you've previously installed from source?

dusty-nv, 2021-12-16:
> Are you asking rosdep to avoid installing packages available throught `apt` that you've previously installed from source?

Not exactly, I believe rosdep already knows which packages that have been installed from source.

So the openCV ones are specified with `--skip-keys` because I manually installed an OpenCV 4.x deb that I compiled (with support for CUDA) and the package names are slightly different or something, and I needed to specify those `--skip-keys` so it didn't install the Ubuntu apt openCV 3.x packages.

Otherwise I use it for problematic packages that it keeps trying to install via apt but makes errors.  Sometimes these have already in fact been installed but ROS doesn't pick them up for some reason.

vectorcrumb, 2021-12-17:
Okay, I'll keep that in mind. Thank you!

closed 2021-12-17 by vectorcrumb