dusty-nv/jetson-containers #115: Cannot build docker: error adding seccomp filter rule ....

y-furuta-hq, 2021-12-17:
I tried to build docker-distro foxy with slam, but could not.
The error code below.
by the influence of  the docker version?

■condition
・Jetson nano 4GB
・Jetpack 4.6
・docker 20.10.7

■error code
$./scripts/docker_build_ros.sh --distro foxy --with-slam
.....
skip
.....
Step 10/29 : RUN locale-gen en_US ....
 ---> Running in 5d35...
failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: error adding seccomp filter rule for syscall clone3: permission denied: unknown

vectorcrumb, 2021-12-17:
I think this is related to #108. Apparently there's an error with the Docker runtime, so you have to patch it to a previous release. The TLDR is:

```bash
wget http://launchpadlibrarian.net/511874292/docker.io_19.03.6-0ubuntu1~18.04.3_arm64.deb
sudo apt install containerd=1.5.2-0ubuntu1~18.04.3
sudo apt install ./docker.io_19.03.6-0ubuntu1_18.04.3_arm64.deb
sudo apt-mark hold docker.io containerd
```
Which gets a previous docker.io, downgrades containerd, downgrades docker.io and finally prevents both packages from being updated during apt upgrade calls.

y-furuta-hq, 2021-12-17:
Thank you all, vectorcrumb.
It's got settled.

closed 2021-12-17 by y-furuta-hq