dusty-nv/jetson-containers #121: Error on build Docker ML all

Fabio81x, 2022-01-21:
Hello
I was tring to compile and run from jetson-container using

./scripts/docker_build_ml.sh all        # build all: l4t-pytorch, l4t-tensorflow, and l4t-ml

But I got following error

`Traceback (most recent call last):
  File "setup.py", line 12, in <module>
    import torch
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 195, in <module>
    _load_global_deps()
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 148, in _load_global_deps
    ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)`

`OSError: libcurand.so.10: cannot open shared object file: No such file or directory`

`The command '/bin/sh -c git clone -b ${TORCHVISION_VERSION} https://github.com/pytorch/vision torchvision &&     cd torchvision &&     python3 setup.py install &&     cd ../ &&     rm -rf torchvision &&     pip3 install --no-cache-dir "${PILLOW_VERSION}"' returned a non-zero code: 1
`

Did someone noticed similar issue? How to resolve?
I'm running on Tx2 Devboard

Package: nvidia-jetpack
Version: 4.6-b199

K.R.

dusty-nv, 2022-01-21:
Hi @Fabio81x, did you set your default docker runtime to nvidia?   https://github.com/dusty-nv/jetson-containers#docker-default-runtime

If not, you may get the error above.

Fabio81x, 2022-01-21:
Hello
yes I followed what writed.

Could be runned those scripts for JetPack 4.6 or there is some limitations? My setup is
- Tx2 developend board
- Reflashed all with Nvidia SDK Manager --> Jetpack 4.6 rev3

After that I start with the wiki

FYI: I tried also to "update" the pip wheels for torch (found in JetsonZoo page), inside Dockerfile.pytorch, with the last link (v1.10.0) but nothing changes, it stop in the same point.

I also searched for the lib inside my system

`jetson-tx2:~$ ls -l /usr/local/cuda/lib64/ | grep libcur`
`lrwxrwxrwx 1 root root        15 mar  1  2021 libcurand.so -> libcurand.so.10`
`lrwxrwxrwx 1 root root        23 mar  1  2021 libcurand.so.10 -> libcurand.so.10.1.2.300`
`-rw-r--r-- 1 root root  62698584 mar  1  2021 libcurand.so.10.1.2.300`
`-rw-r--r-- 1 root root  62767380 mar  1  2021 libcurand_static.a`
`

K.R.

closed 2025-02-21 by johnnynunez